Ultimaker/Cura #3906: Cura 3.4 Infill Support Not Working

RedRightArm, 2018-06-08:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

ChrisTerBeke, 2018-06-08:
Please provide a project fie with the models and settings so we can have a look.

diegopradogesto, 2018-06-20:
I was checking the feature and it works for me, maybe is model dependent. @RedRightArm , did you figure out what happened? Is your issue solved?
I will close this ticket since we didn't receive any info from you. If it is still happening, send us the project file (File > Save Project...) and reopen the issue.

closed 2018-06-20 by diegopradogesto

Dukegriff, 2018-07-11:
I am having the same problem when I turn on infill support it starts to slice but then stops part way through.

diegopradogesto, 2018-07-12:
Please, provide project file.