Ultimaker/Cura #3907: Cura crash on startup

PERECil, 2018-06-08:
**Application Version**
3.3.1

**Platform**
Windows 7 64 bit (fr)

**Printer**
Custom (Creality Ender 3), but not relevant to the situation.

**Steps to Reproduce**
Start cura when installed in a folder that contains non-ascii characters. Mine was `C:\Outils\Multimédia\Cura`

**Actual Results**
Cura crashes.

**Expected results**
Cura starts.

**Additional Information**
When moving cura to another folder without accent in the path (no reinstallation whatsoever) it started working like a charm. I suppose that the accent was the cause.

diegopradogesto, 2018-06-20:
As you said is probably related with the non-ascii characters. Can you please send us the log files? We will take a look anyway.
Thanks for reporting.

diegopradogesto, 2018-06-29:
@LipuFei and me were investigating on this and we figured out what happen. There was a problem with encoding the path of the protobuf file that is used to create the socket in Arcus. 
We made a fix in 3.4: eb32b4f2ec639c53c41c95eb42713417ecdadaa7

closed 2018-06-29 by diegopradogesto