Ultimaker/Cura #3910: Creality C10 defaults to printing mid air

joshhodgson, 2018-06-08:
https://github.com/Ultimaker/Cura/blob/dc5a48fb7db2116a47988e338f8b6cdcdfb08b12/resources/definitions/creality_cr10.def.json#L43

Using this profile, the C10 profile moved to the middle of open space before trying to lay down a bottom, obviously resulting in a big mess and a bad print.

Issue seen when using 3DBenchy.

Am I interpreting this correctly as a mistake?

BuilderPiggie, 2018-06-23:
I have the exact same problem. Also tested with Benchy. However, I don't know if it's only on the CR-10. I got the S version.

@diegopradogesto 
He means the first roof layer of Benchy wants to print from the inside to the outside. That of course, causes that all the filament will fall down. It should print from the outer wall to the inside. But that is not happening. My first Benchy printed just fine some months ago, the first layer of the roof looks waaaaaay better than the newer ones.

JohnEdwa, 2018-08-05:
@ianpaschal I think you are misunderstanding slightly, it doesn't have much to do with the CR-10 itself but the defaults settings of Cura (although I'm not sure if the Ultimaker profiles default to lines or concentric and who actually maintains the 3rd party profiles).  
 
As you sure know the [roof of the Benchy has the walls and corners](https://assets.pinshape.com/uploads/image/file/18133/3dbenchy-the-jolly-3d-printing-torture-test-3d-printing-18133.png) around it and it is meant to benchmark bridging. With standard 'Lines' top surface it almost always works as it isn't that large of a distance, but the Cura default of a 'concentric' top surface [tries to start printing it from the middle outwards](https://i.imgur.com/aMUsZT2.png), which means it utterly fails as there is nothing supporting it.

What @joshhodgson is asking is "Shouldn't Cura understand it can't print in mid-air and not do it here, isn't this a bug", and what I'd ask as a follow up is "Shouldn't Cura use 'lines' as the default top surface as it has the benefit of bridging which is rather common requirement with 3D printing?"

ianpaschal, 2018-08-06:
This would be an issue with the Creality defs. I'm not sure why it says it was concentric but the problem is that we don't really maintain them and don't have any Creality machines to test changes on.

It's probably safe to just change it though... I'll ask @Ghostkeeper who was the one who added it.

Ghostkeeper, 2018-09-28:
I have just changed the default for Creality CR10 because we were getting multiple reports that `Lines` is better for skin: https://github.com/Ultimaker/Cura/commit/3127108d008ab5547ba5e3f1cf45b474d2e88b91

closed 2018-09-28 by Ghostkeeper

probonopd, 2018-10-03:
I ran into the same with the CR-10 profile on an Ender-3 and Cura 3.4.1. The roof gets printed in mid air. I don't think this setting is useful for any type of fused filament printer.

Ghostkeeper, 2018-11-29:
I use concentric every time for the bottom layer, and sometimes for printing something that has a hole in the centre, like a tube. It's useful for some prints and not for others, so I'd like to keep it as an option.