Ultimaker/Cura #3912: [3.4 BETA] Backup not working

mpsdskd, 2018-06-09:
Today I updated to 3.4 BETA and wanted to try out the Backup feature (perfect idea btw).
The backup window appears as intended, but clicking the Sign In button does nothing.

**Application Version**
3.4 BETA Linux 64bit

**Platform**
Kubuntu 17.10

**Printer**
Irrelevant (deleted ~/.local/share/cura & started with new config with a completely new ultimaker 3)

**Steps to Reproduce**
- Enable Backup Plugin 
- Extensions 
- Cuar Backups 
- Manage Backups
- Window for Backups appears with blue "Sign In" button
- Cannot click button/clicking does nothing

**Actual Results**
No way to log in/create account

**Expected results**
Sign in window or Create Account window or login text bar

**Additional Information**
[cura.log](https://github.com/Ultimaker/Cura/files/2086917/cura.log)

Only line containing "Backup":
2018-06-09 13:30:35,752 - WARNING - [MainThread] CuraDrive.src.DriveApiService.getBackups [47]: Could not get access token.

ChrisTerBeke, 2018-06-10:
That log line is expected. What should happen is that a browser window opens with the actual sign-in page. Can you provide some screenshots or a video of what is happening? The logs don't tell anything weird afaik. You could also try and see if there's a small web server running on port 32118 after clicking the button (this is used for the login redirect).

mpsdskd, 2018-06-10:
Web Server seems to be active:
` nmap -p 32118 localhost`
```
Starting Nmap 7.60 ( https://nmap.org ) at 2018-06-10 21:39 CEST
Nmap scan report for localhost (127.0.0.1)
Host is up (0.00021s latency).

PORT      STATE SERVICE
32118/tcp open  unknown

Nmap done: 1 IP address (1 host up) scanned in 0.14 seconds
```

[image]
Clicking the `Sign In`-Button does nothing (as stated above)

I just tried this on my other machine, running Kubuntu 17.10 as well, and got the same results regarding this window.

mpsdskd, 2018-06-10:
Result of 'localhost:32118' in Firefox:
[image]

ChrisTerBeke, 2018-06-10:
Mmm, my guess is that the redirect to the browser is not working then. That's a problem with the Python webbrowser module (https://docs.python.org/3/library/webbrowser.html) then on Kubuntu. Do you have a default browser set?

mpsdskd, 2018-06-10:
Default Application is set to "firefox"
Is there a command I can type eg into an IPython-console to test the webbrowser module?

ChrisTerBeke, 2018-06-10:
You can try `python -m webbrowser -t https://google.com`, that should then open Google in your default browser.

For reference, the source code is here: https://github.com/Ultimaker/CuraPluginOAuth2Module/blob/master/OAuth2Client/AuthorizationService.py#L118.

mpsdskd, 2018-06-10:
I had the same idea and edited my comment.
Ipython3 works as well and opens google in firefox
```
In [2]: import webbrowser

In [3]: webbrowser.open("google.de")
Out[3]: True

In [4]: 

```

ChrisTerBeke, 2018-06-10:
Can you try to find the source code in your user folder and change `open_new` to just `open`?

mpsdskd, 2018-06-10:
`~/.local/share/cura/3.4/plugins/CuraDrive/CuraDrive/lib/CuraPluginOAuth2Module/OAuth2Client/`
-> Line 118 -> ~~`open_new`~~ -> `open`

Changing that line does not seem to change anything.

IPython works:
```
In [2]: webbrowser.open_new("google.de")
Out[2]: True
```

mpsdskd, 2018-06-10:
Could I add some useful logging commands into your script on my machine to help you debugging?

ChrisTerBeke, 2018-06-10:
Everything works as expected, except for that single browser open command. As it's a native Python module, I'm not quite sure why it doesn't work. It also doesn't seem to cause an exception (that would crash Cura or at least throw something in the logs, but there's nothing). I could try switching to the Qt desktop services for opening the browser, but I'd rather find a pure Python solution. So far it only seems to cause issues in Kubuntu though....

ChrisTerBeke, 2018-06-10:
Could you try `webbrowser.get().open(....)` instead of `webbrowser.open()`. That seems to fix some issues for other people according to Google.

mpsdskd, 2018-06-10:
It does not seem to change anything, again.

I have no idea about developing cura plugins, how do you debug these on your machine?
I guess you use some kind of IDE, instead of looking into the cura logs all the time.
But Spyder gives me `No module named 'UM'`, so can you point me to some kind of manual of how to set up my device for plugin development? The Uranium Github wiki was not that helpful.

Thanks

ChrisTerBeke, 2018-06-10:
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source should get you started. I use PyCharm a lot myself.

mpsdskd, 2018-06-10:
I hoped I only had to change some pythonpath variable... I don't have the time for all this, sorry :(

ChrisTerBeke, 2018-06-10:
No problem. Let's hope we can find some other way or people to help debug this.

fieldOfView, 2018-06-11:
Other parts of Cura use QDesktopServices to open the browser. Could that be a solution?

ChrisTerBeke, 2018-06-11:
@fieldOfView possibly, but I'm kinda bummed out that a core Python functionality does not work properly, so really wanna figure that out as well.

fieldOfView, 2018-06-11:
something something code standards something doing the same thing in the same way...

mpsdskd, 2018-06-11:
What would be the command for that and what library do I need to import?

I tried to change in `~/.local/share/cura/3.4/plugins/CuraDrive/CuraDrive/lib/CuraPluginOAuth2Module/OAuth2Client/AuthorizationService.py` :

+`from PyQt5 import QtGui, QtCore`

-`webbrowser.get().open("{}?{}".format(self._auth_url, query_string))`

+`QtGui.QDesktopServices.openUrl(QtCore.QUrl("{}?{}".format(self._auth_url, query_string)))`

Which changed nothing at all.

ChrisTerBeke, 2018-06-11:
If both options don't work it starts to look more and more like an OS (configuration) issue instead of a Python/Qt issue. Does a normal system command like `firefox www.google.com` work?

mpsdskd, 2018-06-11:
I don't think that's where the problem lies, as I can call firefox from command line as well as from an IPython console (using webbrowser as well as QDesktopservices).

I also tried adding a call to open google in a browser instead of the call to what ever url you are creating in your plugin and it did not open the browser.

mpsdskd, 2018-06-11:
I just added some logging code into the function in question:

```
    def startAuthorizationFlow(self) -> None:
        """Start a new OAuth2 authorization flow."""
        
        Logger.log("d", "Starting new OAuth2 flow...")
        
        # Create the tokens needed for the code challenge (PKCE) extension for OAuth2.
        # This is needed because the CuraDrivePlugin is a untrusted (open source) client.
        # More details can be found at https://tools.ietf.org/html/rfc7636.
        verification_code = self._auth_helpers.generateVerificationCode()
        challenge_code = self._auth_helpers.generateVerificationCodeChallenge(verification_code)
        
        # Create the query string needed for the OAuth2 flow.
        query_string = urlencode({
            "client_id": self._settings.CLIENT_ID,
            "redirect_uri": self._settings.CALLBACK_URL,
            "scope": self._settings.CLIENT_SCOPES,
            "response_type": "code",
            "state": "CuraDriveIsAwesome",
            "code_challenge": challenge_code,
            "code_challenge_method": "S512"
        })
        
        # Open the authorization page in a new browser window.
        webbrowser.open("{}?{}".format(self._auth_url, query_string))
        #QtGui.QDesktopServices.openUrl(QtCore.QUrl("{}?{}".format(self._auth_url, query_string)))
        test="{}?{}".format(self._auth_url, query_string)
        Logger.log("w", "I called the URL "+test)
        # Start a local web server to receive the callback URL on.
        self._server.start(verification_code)
```

And this is what I get in the log file:
```
2018-06-11 22:13:47,908 - DEBUG - [MainThread] CuraDrive.lib.CuraPluginOAuth2Module.OAuth2Client.AuthorizationService.startAuthorizationFlow [101]: Starting new OAuth2 flow...
2018-06-11 22:13:47,937 - WARNING - [MainThread] CuraDrive.lib.CuraPluginOAuth2Module.OAuth2Client.AuthorizationService.startAuthorizationFlow [124]: I called the URL https://api.ultimaker.com/auth/v1/authorize?client_id=um---------------ultimaker_cura_drive_plugin&code_challenge_method=S512&response_type=code&code_challenge=kzw5xiwD_b9q1jxd4K450-xOvpKZgm4fJg1GnXrJQljoa9CmFoGRtxmmptugkvKiR-URT0pORgVKR5DpfbRI_Q%3D%3D&scope=user.read+drive.backups.read+drive.backups.write&redirect_uri=http%3A%2F%2Flocalhost%3A32118%2Fcallback&state=CuraDriveIsAwesome
```

[cura.log](https://github.com/Ultimaker/Cura/files/2091581/cura.log)

ChrisTerBeke, 2018-06-11:
For some magic reason it still seems to work fine, except either `webbrowser.open` or `QDesktopServices.openUrl`... really lost here

ChrisTerBeke, 2018-06-11:
Does manually opening that URL and then completing the login flow work?

mpsdskd, 2018-06-11:
Oh, wow, that does work :+1: 

So to summarize: I can call browsers using both Qt and Python standard librarys, but obviously both cannot be called by the backup plugin. Weird!

ChrisTerBeke, 2018-06-12:
Let's see if others have it or not, and on how many different systems it occurs. Thanks for trying everything!

edgary777, 2018-07-11:
Hi,
I just ran into this same issue running Cura 3.4.1 on Ubuntu (Ubuntu 18.04 LTS), I managed to solve it the same way **mpsdskd** did.

I'm commenting because this issue is still marked as open and wanted to let the devs know it is still happening to some linux users.

Ghostkeeper, 2018-07-20:
Does a web browser properly open when you go to Help -> Show Online Documentation?

mpsdskd, 2018-07-20:
No, it does not work. And the Download-Button for a new Version does not work as well.

Good idea, why didn't I think of trying that?!

Morgul, 2018-07-22:
Wanting to throw some more information out there: I'm on Linux Mint 19 (Cinnamon Edition), brand new install, and I'm having the same problem. (Online Documentation menu option also doesn't work).

Ghostkeeper, 2018-07-24:
I could not reproduce this when running from source, but I could reproduce this when running from a build. The Cura log said nothing but the AppImage executable put this into cout:

    XPCOMGlueLoad error for file /usr/lib/firefox/libmozgtk.so:
    /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0: undefined symbol: FcWeightToOpenType
    Couldn't load XPCOM.

I don't really know what this means, but Googling it produces this direction in which we can look: https://github.com/ContinuumIO/anaconda-issues/issues/368
Maybe @LipuFei has an idea about this?

Ghostkeeper, 2018-07-24:
Actually the second result on Startpage is this Github ticket: https://github.com/Ultimaker/Cura/issues/3072

That makes this a duplicate.

closed 2018-07-24 by Ghostkeeper

mpsdskd, 2018-10-27:
With 3.5.1 appimage it works for me now.
Just out of curiosity: Does anybody know what the actual fix was?