Ultimaker/Cura #3913: Adaptive Layers causing missing (top?) layers

megablue, 2018-06-10:
**Application Version**
3.3.1

**Platform**
Windows 10 Pro 64bit with latest updates

**Steps to Reproduce**
Enable Adaptive Layers, 0.1 max variation, 0.05 min steps, 50 threshold  

**Actual Results**
last few layers, a total of 0.5mm height was missing from my models. Despite the gcode preview shows the final height is X the actual gcode stopped printing 0.5mm before the model is actually finished.

**Expected results**
the full models' height should be printed instead of 0.5mm was cut off from the model

ChrisTerBeke, 2018-06-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/3293.

megablue, 2018-06-12:
i dont think it is the same issue, as changing the top layers has no effect on the missing heigh.

Ghostkeeper, 2018-06-28:
This indeed doesn't sound like a duplicate to me, since the issue isn't about the skin height but about actual g-code not being executed by the printer.

I can't find anything wrong in the g-code though. It seems like it should execute fine. Are you able to find a snippet of g-code that the printer has trouble executing?

megablue, 2018-07-04:
the last few layers simply missing, there is no gcode for the last few layers, cura simly decided the print is ended and include the end script.

Consti3D, 2021-01-19:
I can confirm the bug.
With adaptive layers switched on my print object has 0.3 layer height on the bottom and 0.1 layer height on the top.
With layer height set to 0.2 and and bottom and top shell height set to 1.2 mm it calculates 6 bottom and top players. Which results in 1.8 mm bottom shell height and 0.6 mm top shell height.