Ultimaker/Cura #3918: 3>4 Beta

hooch68, 2018-06-11:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

The 3.4 Beta stops slicing about 1/2 way through the process. I let it run for 1 1/2 hrs with no progress

smartavionics, 2018-06-11:
This report is about as much use as a chocolate frying pan. You need to fill out the sections in the template provide project files, screen images, etc. If you don't bother to do that, this issue will be ignored for a while and then deleted. Hope this helps!

fieldOfView, 2018-06-11:
Now I want to make a chocolate frying pan.

ChrisTerBeke, 2018-06-11:
That, and it's very similar to https://github.com/Ultimaker/Cura/issues/3871.

closed 2018-06-11 by ChrisTerBeke