Ultimaker/Cura #3921: Adding new filaments/materials to Cura

jpurcelay, 2018-06-12:
Hello there,

I'm looking to add to the software the profiles for the filaments my company produces (FFF world) but is not clear to me where should I put them. My interest is to add them to the material list and to have them available without the need of importing them manually.

If you could tell me where are the files I should tamper with would be very pleased.

Thank you!

ChrisTerBeke, 2018-06-12:
@Appesteijn

Appesteijn, 2018-06-14:
You can make a Pull Request on Cura:
Material and Quality profiles (for the UM3) can be found https://github.com/Ultimaker/Cura/tree/master/resources/quality/ultimaker3

diegopradogesto, 2018-06-20:
@jpurcelay I will close this issue since you already got the answer. I you need more help, please feel free to open again the issue.

closed 2018-06-20 by diegopradogesto