Ultimaker/Cura #3924: End code inherited from Malyan M200 crashes MP Select Mini V2 (Cura 3.3.1)

gnbl, 2018-06-13:
With Cura 3.3.1, the Monoprice Select Mini V2 hits limits due to inheriting "end code" from Malyan 200 that moves to `Y127`, which should be `Y120`:

https://github.com/Ultimaker/Cura/blob/1b3706e172f97812e2b78371303c8f93f7c7ef62/resources/definitions/monoprice_select_mini_v2.def.json#L5

https://github.com/Ultimaker/Cura/blob/1b3706e172f97812e2b78371303c8f93f7c7ef62/resources/definitions/malyan_m200.def.json#L52

Possible solutions: 

- Add correct, individual end code to the printer's definition's `overrides` (`monoprice_select_mini_v2.def.json`):

```
        "machine_end_gcode" : {
            "default_value": "G0 X0 Y120;(Stick out the part)\nM190 S0;(Turn off heat bed, don't wait.)\nG92 E10;(Set extruder to 10)\nG1 E7 F200;(retract 3mm)\nM104 S0;(Turn off nozzle, don't wait)\nG4 S300;(Delay 5 minutes)\nM107;(Turn off part fan)\nM84;(Turn off stepper motors.)"
},
```

- Parse end code and limit to printer maximum values.
- Use inherited code as template and insert printer values.

gnbl, 2018-06-13:
Hi, thanks for getting back.
Of course I can only speak for my printer, where I have changed the end code to 120 via the GUI, which resolved the problem (changes stored in %AppData%\cura\3.3\definition_changes\).
I have not tried other values, and I don't know if this is the exact value (offsets, mechanical tolerances, ...).

Maybe people are not using this "recent" Cura (since Nov 2017)? Maybe nobody uses these profiles but instead use their own? Maybe it does not happen on all models? Maybe people don't notice or think it's supposed to do that?

I just have come across another report on reddit (with video) and that caused me to open this issue:
https://www.reddit.com/r/MPSelectMiniOwners/comments/8qo5xn/why_does_my_printer_make_this_vibration_sound_at/

Searching https://www.reddit.com/r/MPSelectMiniOwners/search?q=127&restrict_sr=1 results in a solution post: https://www.reddit.com/r/MPSelectMiniOwners/comments/8pusul/y_axis_grinding_at_end_of_print/

Ghostkeeper, 2018-07-24:
For information like this we have to trust in our community to get this right. If gnbl says that the end code needs to be changed and that he tested this, I'll trust him on that.

commit 75d8cba8 (2018-07-24): Fix start g-code of Monoprice Select Mini v2

Apparently the original start code caused it to move outside the build volume or beyond the limit in the Y direction.

Fixes #3924.
closed 2018-07-24 by Ghostkeeper