Ultimaker/Cura #3927: Experimental category missing

BagelOrb, 2018-06-13:
**Application Version**
Cura 3.4-beta (and quite some versions before that as well in my recollection)

**Platform**
Linux Mint

**Printer**
UM3 (likely any)

**Steps to Reproduce**
Make any experimental setting visible
Then clear the search field and scroll down to the setting you just made visible

**Actual Results**
Category header is missing.
The setting is just under the last visible category.

**Expected results**
When any setting from the experimental category shows up, the header also shows up.

**Additional Information**
Animated gif:
[image]

diegopradogesto, 2018-06-20:
I checked in Windows and I couldn't reproduce it. @Ghostkeeper @jackha @alekseisasin or somebody else, can you try in Linux?

BagelOrb, 2018-06-20:
I just reproduced it on Windows for Cura 3.1 (my home computer has an old
Cura version)

BagelOrb, 2018-06-20:
Scrap that

BagelOrb, 2018-06-20:
Can't reproduce on Windows either.

jackha, 2018-06-20:
Can't reproduce it. Maybe Cura was in the "Basic" settings view? (The button next to the search bar.)

closed 2018-06-20 by jackha

BagelOrb, 2018-06-20:
I guess it had to do with my config file then. It might have gotten
scrambled due to changing branches all the time or something.