Ultimaker/Cura #3928: Material diameters do not stay as set

brianzhang-git, 2018-06-13:
**Application Version**
3.2.21

**Platform**
Windows 10

**Printer**
Taz 5

**Steps to Reproduce**
When creating a new material or editing a custom material, set the diameter to 1.75 or 2.85 mm and set other options. Close the materials dialog and re-open.

**Actual Results**
The material diameter has been reset to 0.00 mm.

**Expected results**
The material diameter stays at the previously set diameter.

**Additional Information**
When opening the custom material XML, there is nothing set for <diameter>. I can add in <diameter> manually and it will take/show in Cura. However, if I then try to modify that diameter in the Cura dialog, it removes the material from the list. It is still the active material, but is no longer visible. In the file, <diameter> has moved forward, in front of <density>. If I attempt to re-import the profile, Cura tells me it has imported successfully but I cannot see the profile still. 

Overall, a series of rather complex problems with the material profiles. I think that the root cause is something with the reading/writing of <diameter>, but I'm not very sure.

ChrisTerBeke, 2018-06-14:
3.2.21 is not a Cura version that Ultimaker ships...

ChrisTerBeke, 2018-06-14:
https://github.com/Ultimaker/Cura/issues/3930

closed 2018-06-14 by ChrisTerBeke