Ultimaker/Cura #3931: [3.4 BETA] Z-hop not working on retract while combing

PKav89, 2018-06-14:
**Application Version**

3.4 BETA

**Platform**

Windows 10

**Printer**

Ultimaker 2 Extended

**Steps to Reproduce**

Slice with following settings:
Combing: All
Max combing distance with no retraction: 10mm (any value will do)
Z-hop: 0.1 (any value will do)

Z-hop only over printed areas: No

**Actual Results**

Z-hop happens on the first and top layer, but does not happen on other layers and scratches over just printed layer.

**Expexted Results**

Z-hop happens every time the retraction happens.

**Additional Information**

I was waiting for "Max combing distance with no retraction" setting for so long! Now at last it's here, but works not as expected. It seems like a realy easy fix, so, please, fix it and I will be completely happy with Cura.

Here is G-Code from the first layer with Z-Hop working as expected:

```
G1 X6.263 Y19.891 E127.94988
G1 X6.784 Y19.588 E128.04752
G1 X7.852 Y19.092 E128.23828
G10
G1 F2400 Z0.37
G0 F12000 X11.135 Y109.492 Z0.37
;TYPE:WALL-INNER
G1 F2400 Z0.27
G11
G1 F600 X11.496 Y109.426 E128.29773
G1 X11.701 Y109.363 E128.33247
G1 X11.921 Y109.275 E128.37086
```

And here is the G-Code from the upper layer with retraction but without Z-Hop

```
G1 X40.811 Y120.031 E7691.37008
G0 F12000 X40.619 Y120.087
G0 X40.496 Y119.665
G10
G0 X40.48 Y119.607
G0 X70.022 Y116.208
G0 X70.12 Y116.194
;TYPE:WALL-INNER
G11
G1 F1080 X70.19 Y116.496 E7691.38868
G1 X70.293 Y116.762 E7691.4058
```

Here is my project file:
[Panel_Holder_x6.curaproject.zip](https://github.com/Ultimaker/Cura/files/2100967/Panel_Holder_x6.curaproject.zip)

smartavionics, 2018-06-14:
Hello @PKav89 , that's correct, you don't get z-hop during combing. That's how it's designed to be (not by me, that's how it's always been). If you want z-hop, turn off combing and just use normal retract+hop moves.

PKav89, 2018-06-14:
Hello @smartavionics , Combing helps to keep leaking plastic within the perimeter and eliminates strings. Even ABS leaks a little between G0 movements, causing small "hairs" on the shell. To eliminate this I had to set G0 speed to 200 mm/sec, it's OK for Ultimaker and deltas, but too fast for direct-drive extruders. But if you think it's not important, please close the issue. Retrect while combing is still great improvement in Cura 3.4 so it won't generate any weird G0 moves in "No skin" combing mode and won't put strings on the first layer in "All' mode.

smartavionics, 2018-06-14:
I can't close the issue, only you or a Cura "guardian" can do that. But if you think there is a strong argument for allowing z-hop during combing then press on and have a conversation with the developers about that. I would be happy to make further code changes but I'm not going to bother unless the Cura developers are willing to accept the changes. My guess is that because the UM printers are so slow in Z movement that they try to avoid z-hop at all times. Personally, I don't have a problem with it as my deltas can z-hop as quick as z/y movement.

jackha, 2018-06-28:
Hi @PKav89. Thanks for the report and discussion. What we want to know is why you actually want Z-Hop + Combing? As far as I'm concerned, if you're able to comb that's better than z hopping. So you're getting "hairs" on the shell while combing? Is that a model with thin parts? I would like to see a project file to get more insight of what's happening here.

PKav89, 2018-06-28:
Hi @jackha.
I've managed to get rid of the "hairs" by tweaking movement settings. It still scratches infill with a loud sound while combing, but that's not a big problem. Please close this issue.

jackha, 2018-07-02:
Glad you got rid of the hairs :-)

closed 2018-07-02 by jackha