Ultimaker/Cura #3932: cura crashes while adding fdm printer or printer based on it

bzed, 2018-06-14:
**Application Version**
3.2, 3.4, master

**Platform**
debian testing/unstable

**Printer**
custom fdm, cr10 S4 and some other random picked printers based on fdm

**Steps to Reproduce**
Add new printer -> choose a non-ultimaker one -. OK

**Actual Results**
cura crashes, output is pasted below

**Expected results**
cura should not crash and add the printer :) Adding Ultimaker printers works just fine.

**Additional Information**

```
2018-06-13 21:47:14,500 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [53]: Loading container custom_extruder_1
2018-06-13 21:47:14,527 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_1] added.
2018-06-13 21:47:14,549 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_1 #2_settings] added.
2018-06-13 21:47:14,549 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_1 #2_user] added.
2018-06-13 21:47:14,550 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[custom_extruder_1 #2] added to [Custom FDM printer] at position [0]
2018-06-13 21:47:14,550 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [custom_extruder_1 #2] has already been added to this stack [Custom FDM printer]
2018-06-13 21:47:14,554 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_1 #2] added.
2018-06-13 21:47:14,554 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [53]: Loading container custom_extruder_2
2018-06-13 21:47:14,583 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_2] added.
2018-06-13 21:47:14,605 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_2 #2_settings] added.
2018-06-13 21:47:14,605 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_2 #2_user] added.
2018-06-13 21:47:14,606 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[custom_extruder_2 #2] added to [Custom FDM printer] at position [1]
2018-06-13 21:47:14,606 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [custom_extruder_2 #2] has already been added to this stack [Custom FDM printer]
2018-06-13 21:47:14,610 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_2 #2] added.
2018-06-13 21:47:14,610 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [53]: Loading container custom_extruder_3
2018-06-13 21:47:14,639 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_3] added.
2018-06-13 21:47:14,661 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_3 #2_settings] added.
2018-06-13 21:47:14,662 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_3 #2_user] added.
2018-06-13 21:47:14,662 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[custom_extruder_3 #2] added to [Custom FDM printer] at position [2]
2018-06-13 21:47:14,662 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [custom_extruder_3 #2] has already been added to this stack [Custom FDM printer]
2018-06-13 21:47:14,666 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_3 #2] added.
2018-06-13 21:47:14,666 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [53]: Loading container custom_extruder_4
2018-06-13 21:47:14,698 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_4] added.
2018-06-13 21:47:14,723 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_4 #2_settings] added.
2018-06-13 21:47:14,724 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_4 #2_user] added.
2018-06-13 21:47:14,724 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[custom_extruder_4 #2] added to [Custom FDM printer] at position [3]
2018-06-13 21:47:14,724 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [custom_extruder_4 #2] has already been added to this stack [Custom FDM printer]
2018-06-13 21:47:14,728 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_4 #2] added.
2018-06-13 21:47:14,728 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [53]: Loading container custom_extruder_5
2018-06-13 21:47:14,759 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_5] added.
2018-06-13 21:47:14,779 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_5 #2_settings] added.
2018-06-13 21:47:14,780 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_5 #2_user] added.
2018-06-13 21:47:14,781 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[custom_extruder_5 #2] added to [Custom FDM printer] at position [4]
2018-06-13 21:47:14,781 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [custom_extruder_5 #2] has already been added to this stack [Custom FDM printer]
2018-06-13 21:47:14,785 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_5 #2] added.
2018-06-13 21:47:14,785 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [53]: Loading container custom_extruder_6
2018-06-13 21:47:14,812 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_6] added.
2018-06-13 21:47:14,832 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_6 #2_settings] added.
2018-06-13 21:47:14,833 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_6 #2_user] added.
2018-06-13 21:47:14,833 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[custom_extruder_6 #2] added to [Custom FDM printer] at position [5]
2018-06-13 21:47:14,834 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [custom_extruder_6 #2] has already been added to this stack [Custom FDM printer]
2018-06-13 21:47:14,838 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_6 #2] added.
2018-06-13 21:47:14,838 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [53]: Loading container custom_extruder_7
2018-06-13 21:47:14,865 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_7] added.
2018-06-13 21:47:14,885 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_7 #2_settings] added.
2018-06-13 21:47:14,886 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_7 #2_user] added.
2018-06-13 21:47:14,886 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[custom_extruder_7 #2] added to [Custom FDM printer] at position [6]
2018-06-13 21:47:14,887 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [custom_extruder_7 #2] has already been added to this stack [Custom FDM printer]
2018-06-13 21:47:14,891 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [custom_extruder_7 #2] added.
[... 87 lines cut ...]
2018-06-13 21:47:15,410 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/local/lib/uranium/plugins/LocalContainerProvider/LocalContainerProvider.py", line 63, in loadContainer
2018-06-13 21:47:15,410 - ERROR - [MainThread] UM.Logger.logException [85]:     container.deserialize(f.read())
2018-06-13 21:47:15,410 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/usr/local/lib/python3/dist-packages/UM/Settings/InstanceContainer.py", line 496, in deserialize
[... cut ...]
2018-06-13 21:47:15,415 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "/usr/local/lib/python3/dist-packages/cura/Machines/ContainerNode.py", line 43, in getContainer
2018-06-13 21:47:15,415 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     raise RuntimeError("Failed to lazy-load container [%s], cannot find it" % container_id)
2018-06-13 21:47:15,415 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: RuntimeError: Failed to lazy-load container [low], cannot find it
```

ChrisTerBeke, 2018-06-14:
Creating a custom FDM printer works fine for me on 3.4 Beta...

Appesteijn, 2018-06-15:
Do you have a custom profile: 'Low' ?

Failed to lazy-load container [low], cannot find it