Ultimaker/Cura #3935: Ultimaker 2 prints only 500MB of Gcode file

msiccha, 2018-06-15:
**Application Version**
Cura 3.3.1

**Platform**
Windows 10

**Printer**
Ultimaker 2+ (upgraded UM2)

**Steps to Reproduce**
Generate a gcode file larger than 500MB and print it

**Actual Results**
Print result is incomplete, the last X levels are not printed

**Expected results**
Full print

**Additional Information**
I cannot attach a project file as this would be very large, but there is a photo of an example icomplete print of a large model.
[image]

Vandrasc, 2018-06-15:
Hi @msiccha, does the gcode looks complete?

[Save to File > Save as type Gcode]

And, did you tried to load the gcode back into Cura and verify the layer view? Does your model look complete? Thanks!

msiccha, 2018-06-15:
Hi @Vandrasc ,
I cannot look at the gcode at the moment (no editor installed that can view files that large) but if I reload the model from the GCODE file to Cura it is complete and the layer view shows nothing suspicious.

ianpaschal, 2018-07-10:
If you are seeing the whole gcode in Cura after slicing and reloading the gcode then this appears to be a problem with the printer. Normally that is very hard for us to diagnose but since this is a UM printer we can try to take a look. Would be useful to have your model and project file when possible.

Ghostkeeper, 2018-08-24:
I asked one of our firmware engineers and he hadn't heard of this problem before. He thinks that you might have a good chance of getting an answer from the [Ultimaker forum](https://community.ultimaker.com/forum/109-firmware/).

For now this looks like a problem in the UM2 firmware rather than in Cura. Alternatively, you could also shoot in an issue at [their repository](https://github.com/Ultimaker/UM2.1-Firmware/issues). In any case, we'll close it here.

closed 2018-08-24 by Ghostkeeper

ckielstra, 2018-08-24:
The issue is already closed here, but as a comment from the firmware team: 
We have tested the UM2(+) with files up to 1.3Gb working flawlessly. Files larger than 500Gb are used by many people without problems.

Reading all information, my suspicion is something odd with your memory card. There are many brands and models of memory cards, perhaps you have an incompatible card. What is the brand/model you are using?

Workaround: use another brand/model memory card.

DDDirk, 2018-08-24:
Of format the card. This solved a similar issue for me once.