Ultimaker/Cura #3936: Spurious "file reload?" notifications posted

smartavionics, 2018-06-15:
**Application Version**

master branch

**Platform**

Linux

**Printer**

Kossel Mini

**Steps to Reproduce**

Load a gcode file to view, delete that and then load model(s), slice and save gcode using the same filename as was viewed earlier.

**Actual Results**

Unexpected notifications appear offering to reload the gcode file that was previously viewed.

**Expected results**

Notifications should not appear.

**Additional Information**

See attached screenshot.

I have also seen many of these notifications appearing when I have a part duplicated and I update the part's STL file. Only one need be posted in that situation, not one for each of the instances of the part.

[image]

Vandrasc, 2018-06-15:
Ah, @Ellecross, look at this :) What was the decision regarding the unnecessary messages and Reload button?

diegopradogesto, 2018-06-20:
For devs.: CURA-5478

Ghostkeeper, 2018-08-02:
This issue was fixed yesterday, so you shouldn't get the message twice any more, the reload button should work properly, the message should disappear when you reload, and when you save a file from Cura on top of a currently opened file you should not get a message to reload.

closed 2018-08-02 by Ghostkeeper

smartavionics, 2018-08-15:
Back on this again. Current master branch posts multiple notifications when stl file is updated. This is with just one update to a single instance of the model.

[image]

smartavionics, 2018-08-16:
This needs to be reopened.

Also, I notice today that the gcode files that I save are now appearing the recent file list which is not what I want at all. Surely that's a regression?

Ghostkeeper, 2018-10-15:
I'm unable to reproduce it at this time.

[image]

Maybe it was fixed again in the meanwhile?

I do remember that the g-codes in the recent file list was a regression at some point.