Ultimaker/Cura #3938: Lower print speed at dissimilar material interface

AbeFM, 2018-06-15:
New islands printed on PVA can be troublesome.

Previous suggestions for "caging" the first layer or two on supports generated little interest.

One could treat the first layer or two on top of support interface as one does the first layers on the print bed - lower speeds, different extruder temps.

This would allow full speed printing on most of the model where only a few lines actually need to print slowly.

**Application Version**
3.3.1

**Platform**
Win 10

**Printer**
Dual extruder custom

AbeFM, 2018-06-18:
...I bring this up because I can print PVA-only objects at 60+ mm/s without any issue. But as soon as I use it as an interface material, I have to add hours upon hours to my print so that everything prints at 10mm/s for the <<1% of the time where I have interface printing.

I can print the upper layer at different speeds (not temps!) by using ceiling/floor settings on support, but the issue is dissimilar materials.

jackha, 2018-06-28:
Hi @AbeFM . Thanks for your report.

What do you mean by "troublesome"? Is the PVA not sticking to the bed? The new islands not sticking to the PVA? Can you share a project file with us to show what you mean?

AbeFM, 2018-06-28:
PVA sticks questionably to any surface. Not unusable, but not like other materials. Any dissimilar materials have this issue, and one could make the argument that for fine enough structures, even similar materials are rough if you're not laying down enough hot filament to melt whatever it is sticking to.

That said, the issue I'm addressing here is new islands. It is the worst with sharp points, but any new island would benefit from slower printing when beginning, and then you can ramp up speed once it is set. These interfaces make up well under 1% of the total model but the speed reduction is applied everywhere, so it's a loss-multiplication of like 100:1!

Check:
https://github.com/Ultimaker/Cura/issues/3894
Or even the end of:
https://community.ultimaker.com/topic/21457-support-for-sharp-points-facing-down/

Would you still like a project file? I can whip something  up.

jackha, 2018-07-02:
Ah! "Support Floor" is meant for this, you enable it by checking "Enable Support Floor". It does not change the speed, but you can set the density (which defaults to 100%) and the thickness.
Let me know if this helps.

AbeFM, 2018-07-02:
Almost! It's half of it, but a good catch cause I don't even think of it.

The support floor controls print speed of first layers of support printed
on top of the part (not the bed. I asked for that but was told no one would
want to have fine control over the first layers of support on print bed,
even if you need that everywhere else).

That aside, the issue I have is new islands printed on support roof
(ceiling?) of the support interface is printed according to the settings
but the first layer of your printed object to go down on that support is
printed at full speed, with no settings just for the interface.

That is what I would like to see.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

jackha, 2018-07-03:
Ah so I misunderstood. Your problem is that new islands of PLA falling down on PVA that was printed too fast, right? There is actually a Support Roof setting! Let me know if that helps.

AbeFM, 2018-07-03:
To be explicit:

The top layer of the roof, made out of PVA, prints well.

Next layer I print the first bit of island of PLA on top of the existing
support roof.

I would like that first layer of PLA to print more slowly than it
ordinarily would because it is on PVA.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

jackha, 2018-07-04:
I just discussed with the team. Adjusting bottom skin settings is actually already in our backlog ^^.
Devs: CURA-5411

AbeFM, 2018-07-04:
Awesome!

Let me know if there's any burning questions otherwise I'll leave you fine
folks to it.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Ghostkeeper, 2018-08-15:
To be clear, the feature in our planning is to adjust the speed and pattern of the bottom layer(s) of skin, regardless of whether that bottom layer is resting on support, the build plate or air. It's not exactly what was requested here but it would help towards Abe's goal.

AbeFM, 2018-08-15:
It certainly would work. It's unnecessary as you're slowing down all bottom
surfaces when you're only using that slowdown in a few places, but it would
be helpful, and likely solve other issues as well.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes