Ultimaker/Cura #3939: [3.3.1] critical flat top/bottom broken shell issue

St007, 2018-06-16:
**Application Version**
3.3.1

**Platform**
Win10

**Printer**
All printers

**Steps to Reproduce**
Slice model with default settings https://www.thingiverse.com/thing:2524462

**Actual Results**
Shell is interrupted with infill.

**Expected results**
Shell + outside perimeter must form continues object

**Additional Information**
Attachments: schematic of issue, broken printed object (it is really easy to broke because of this issue), compare with Simplify3D (how it must be)
[image]
[image]
[image]

smartavionics, 2018-06-16:
Hi, the only way I can recreate what you are showing above is to have a few mm of skin removal width and zero skin expansion. If you reduce the skin removal width it won't zap so much skin. In fact, for this model you probably want the skin removal width to be zero and have a non-zero amount of skin expansion.

[image]

St007, 2018-06-16:
smartavionics, thank you for suggestion
After correction, now I have same result as with Simplify3D, however these settings are hidden by default/ Default value for Skin Removal Width equals shell width which leads into issue of this topic. So, for new users who do not know which settings affecting such result, the behavior will look like critical issue. I think Skin Removal Width must be zero by default. So, my suggestion is to keep open with issue report and wait for Ultimaker dev team response.
[image]

smartavionics, 2018-06-16:
I agree, leave it open so that a discussion can be had around the default value of skin removal width. Personally, I would tend to agree with you that it should default to zero or a low value.

closed 2018-06-27 by St007