Ultimaker/Cura #3943: Cura 3.3.1 does not start on win10 Lenovo t430

kaibode, 2018-06-17:
**Application Version**

Cura 3.3.1 final
**Platform**

Windows 10 latest
**Printer**

none, yet, initial startup failed
**Steps to Reproduce**

Cura starts, splashscreen, loading machines etc, then white windows opens with disclaimer. Then long wait, Windows then declares the program cura.exe not working and only offers close program as option
**Actual Results**

cura.exe crashes
**Expected results**

no crash
**Additional Information**

I added log-files
[stderr.log](https://github.com/Ultimaker/Cura/files/2109022/stderr.log)

[cura.log](https://github.com/Ultimaker/Cura/files/2109020/cura.log)

kaibode, 2018-06-18:
I went back in releases, and the last one that works on my PC is version 2.3.1. 64bit Is there anything that happened between 2.3.1 and 2.4?

Kai.

jackha, 2018-06-28:
Hi @kaibode . Thanks for your report.

In Cura 2.5 we added a new Layer View and added OpenGL version detection, so I would suspect that it broke from that moment on. From OpenGL 4.1 and higher you'll see the "new" 3D layer view, while on lower versions it shows the "legacy" layer view. Well, that's supposed to happen.

I can see in the logs that your graphics card is OpenGL 4.0.0 on an Intel HD Graphics 4000. Do you have the most recent graphics drivers? That would be my main suspect.

kaibode, 2018-06-28:
Hello,

Indeed, after installing new graphics drivers, first for Intel built in, then for the Nvidea, everything is working now. i also upgraded to 3.4 just to be sure.

You rock, Jackha, thank you!!!

Kai.

P.S.: maybe include in the readme, that before bothering support one should upgrade to latest graphics drivers.

alekseisasin, 2018-06-29:
@kaibode , probably we need add to Frequently asked questions (FAQ).

closed 2018-06-29 by alekseisasin