Ultimaker/Cura #3944: [3.3.1] appdata.xml contains invalid XML entities

onitake, 2018-06-17:
**Application Version**
Cura 3.3.1 (and all previous versions)

**Platform**
Linux

**Printer**
Not applicable

**Steps to Reproduce**
Run `xmllint` on cura.appdata.xml

**Actual Results**
`xmllint` complains about unescaped entities:
```
cura.appdata.xml:29: parser error : EntityRef: expecting ';'
page">https://ultimaker.com/en/products/cura-software?utm_source=cura&utm_medium
                                                                               ^
cura.appdata.xml:29: parser error : EntityRef: expecting ';'
r.com/en/products/cura-software?utm_source=cura&utm_medium=software&utm_campaign
                                                                               ^
```

**Expected results**
No errors.

**Additional Information**

This issue appeared because Linitian includes a new check for appstream XML validity since 4.1.4, but the file was incorrect before.

Patch to fix this issue:
```
--- a/cura.appdata.xml
+++ b/cura.appdata.xml
@@ -26,6 +26,6 @@
   <screenshots>
     <screenshot type="default" width="1280" height="720">http://software.ultimaker.com/Cura.png</screenshot>
   </screenshots>
-  <url type="homepage">https://ultimaker.com/en/products/cura-software?utm_source=cura&utm_medium=software&utm_campaign=resources</url>
+  <url type="homepage">https://ultimaker.com/en/products/cura-software?utm_source=cura&amp;utm_medium=software&amp;utm_campaign=resources</url>
   <translation type="gettext">Cura</translation>
 </component>
```

onitake, 2018-06-18:
:+1: Here you go.

ChrisTerBeke, 2018-06-18:
Thanks! Closing this in favor of the PR.

PR #3948 (2018-06-19): Fixed incorrect appstream data XML.
Fixes: #3944
commit 42337f60 (2018-06-19): Fixed incorrect appstream data XML.
Fixes: #3944
closed 2018-06-18 by ChrisTerBeke