Ultimaker/Cura #3945: [3.3.1]on Mac creates gaps on layers and on Windows doesn't

Axoluto, 2018-06-17:
**Application Version**
Cura 3.3.1

**Platform**
Mac OSx El Capitan 10.11.6

Hi all,
i would appreciate some help to confirm the facts i noticed.
If i open the uploaded file to cura on Mac OSx El Capitan 10.11.6, Cura creates lacks of layers in various places of the object, i tried two different versions of Cura and it did the same. So i asked a friend of mine to open it on Windows with Cura 3.3.1 and it sliced great with no errors.
[T800_Base_Supported_v5.3mf.zip](https://github.com/Ultimaker/Cura/files/2109424/T800_Base_Supported_v5.3mf.zip)

I have a photo of the problem:
[image]

Is it possible to someone try to open it and reply the results, thanks. 

*

AbeFM, 2018-06-18:
I would suggest that you really beat that up - move it around the bed, tweak settings, etc. I've had missing layers like this before on Windows, and often moving or fixing the model would change it.

That said, generally I will get a single missing layer where it looks like you have several. It might make sense to have him not just slice the model, but to use your entire settings file.

Axoluto, 2018-06-18:
Thanks. I just solved it. Just changed the "Surface mode" from both to normal.

closed 2018-06-18 by Axoluto

Ghostkeeper, 2018-07-31:
> I've had missing layers like this before on Windows, and often moving or fixing the model would change it.
> 
> That said, generally I will get a single missing layer where it looks like you have several.

We have a known bug where sometimes a layer would get rendered in layer view at position Z=0. That's only layer view though. The actual g-code is good. And a re-slice usually fixes it.

> Just changed the "Surface mode" from both to normal.

"Both" should logically always print more than "normal", right? Still some weirdness going on, I'd say. But I can't reproduce the issue myself.

AbeFM, 2018-08-01:
I'd have to see it to believe it.

In my experience, a print fails, I review g-code in Cura and find missing layers as shown above.