Ultimaker/Cura #3949: Cura 3.3.1 doesn't work

Lucacenci, 2018-06-18:
after the Upgrado of the Cura software the printer is not working, (invalid code), some help?
thank you

ChrisTerBeke, 2018-06-18:
Please fill out the issue template, attach logs and/or screenshots or anything else to clarify your issue.

Lucacenci, 2018-06-18:
[image]
this is the message, all the program made with new Printer is not working

ChrisTerBeke, 2018-06-18:
And the explanation on ultimaker.com/ER26 doesn't help you further? Seems like you're slicing for the wrong machine in Cura?

Lucacenci, 2018-06-18:
yes but thr printer is correct

ChrisTerBeke, 2018-06-18:
Can you share a project file (in Cura click file -> safe as project)?

ianpaschal, 2018-06-28:
More than a week without response. If @Lucacenci gets back to us we can reopen otherwise this is going to be forgotten about.

closed 2018-06-28 by ianpaschal

Ghostkeeper, 2018-08-09:
Most likely this is the result of accidentally enabling "Origin at Center".

Could you go into the Machine Manager, click "Machine Settings" and make sure that "Origin at Center" is disabled?