Ultimaker/Cura #3951: Material-Based Fan Speed Not Applied

wilriker, 2018-06-18:
**Application Version**
3.3.1

**Platform**
Arch Linux

**Printer**
Custom FDM

**Steps to Reproduce**
1) Define a material and set fan speed to 30%
2) Select material

**Actual Results**
Fan speed for slicing stays at 100%

**Expected results**
Fan speed for slicing should be change to 30%

**Additional Information**
It works correctly for all other material-definable settings, i.e. temperatures and rectraction settings, just not for fan speed.

AbeFM, 2018-06-18:
I think this is one of those cases for what happens first. I've had, and mentioned, similar issues (like retract) which are more effected by the printer than the material, or at least the two effects are interdependent.

jackha, 2018-06-28:
Hi @wilriker . Thanks for the report. However, I can't reproduce this on 3.3.1 or the new 3.4.

If you set the fan speed, does it instantly change in the settings screen as well? (Open Cooling and set Setting visibility to at least "Advanced" first).

wilriker, 2018-06-28:
Hi @jackha. After writing up a whole new very verbose way of reproducing this I just found the real cause of my issue:

My printing profile has the `Fan Speed` set to `Calculated` which in its title text says
```
=100.0 if cool_fan_enabled else 0.0
```
If I create a new profile that does not have this setting it works as intended. Sorry for the hassle.

But my follow-up question: how can I get rid of this setting in my profile without creating a new one? I think I clicked on this little (i) icon next to the setting that recommended me changing this from an absolute value to a calculated one - and I thought that this would do what I expected but actually it is the other way round. :man_facepalming:

jackha, 2018-06-28:
Ahh that makes sense. Glad you found it. How did you create this profile in the first place? Functions can only be added by manually changing the quality files right? (Quality) Profiles indeed go over material profiles.

wilriker, 2018-06-28:
I created this profile by duplicating the built-in `Draft Quality` profile. Then I adjusted my settings via the settings side-bar and saved the changes to this profile.
I did not add the above function manually it just describes what's the meaning of `Calculated` - and that one was recommended to me by the UI. So if you change the value of `Fan Speed` manually a little i-icon will appear next to the input field telling you that this setting usually is calculated but currently set to an absolute value and one can click the icon to restore it to be calculated.

But now I cannot find a way to undo this to basically remove any `Fan Speed` setting from the quality profile. I don't mind editing config files I just need a hint where to find it.

wilriker, 2018-06-28:
Just found the file. On Linux it was at
```
~/.local/share/cura/3.3/quality/custom_extruder_1_%232_<profilename>.inst.cfg
```

Removed the according line there and now it works as expected. :+1:

closed 2018-06-28 by wilriker

jackha, 2018-07-02:
Glad to hear you found it!

SkullKill, 2020-05-11:
to fix this, i edit all the files in 
C:\Users\USER\AppData\Roaming\cura\4.5\quality_changes\*.cfg

and removed all lines that contained,
"cool_fan_speed = "