Ultimaker/Cura #3952: [3.3.1] add a way to change flow % for spiralized prints after bottom layers

hantaski, 2018-06-18:
**Application Version**
3.3.1

**Platform**
win 10 64bit

**Printer**
any

**Steps to Reproduce**
select spiralized outer contour mode

**Actual Results**
flow stays at 100% after bottom layers are finished

**Expected results**
flow should be different after bottom layers

**Additional Information**
Couldn't find a way to add a feature request, so doing it this way. As many know, spiralized or vase mode prints are better when filament flow is increased (for example to 150%) but that causes bottom layers to be bad. So I am asking to add a feature, that when spiralized mode is active, it would be possible to choose different flow rate for a actual spiral and a different one for a bottom layers. If this is already somehow possible, then sorry about this.

smartavionics, 2018-06-18:
For spiral prints, I up the line width to 0.6mm when using a 0.4mm nozzle, that's exactly the same in terms of amount of plastic extruded as asking for a flow of 150%,

smartavionics, 2018-06-28:
> @smartavionics don't you have to square it to calculate relative flows? 0.60.6 / (0.40.4) = 2.25, so that's a flow rate of 225%.

No, because the area of the line is layerHeight * lineWidth. When you increase the line width, the layer height stays the same so it's not a squared relationship. Doubling the line width, doubles the flow.

Ghostkeeper, 2018-08-10:
I think upping the wall line width is a fine solution for this. Increasing the flow is only applicable if you specifically want overextrusion or underextrusion to compensate for some other effect. But here you just want to have stronger walls so a thicker line width is a very good way to achieve that.

closed 2018-08-10 by Ghostkeeper