Ultimaker/Cura #3953: Monitor - Active Print - Printing Time not showing after print is ready - Cura 3.3.1

waterwin, 2018-06-18:
**Application Version**
Cura 3.3.1

**Platform**
Windows 10 64 bit

**Printer**
Ultimaker Original

**Steps to Reproduce**
Do a print 

**Actual Results**
Print is ok but Printing time is empty

**Expected results**
Software should show the time it took to print
[image]

**Additional Information**
When the print start a count down timer is showing starting from -2.00.00 and counting down which has no relation to the estimated printing time before starting the print via usb

alekseisasin, 2018-06-29:
@waterwin , we fixed the starting time should be fixed. Please check the latest Cura version.

ianpaschal, 2018-07-10:
^Haven't heard back so I will consider this a "yes"

closed 2018-07-10 by ianpaschal