Ultimaker/Cura #3954: Unexpected layer shift

dventu, 2018-06-19:
**Application Version**
 3.3.1

**Platform**
 WIN 10 Pro

**Printer**
 Custom

**Steps to Reproduce**
 Printer running tmc 2130. print at any speed

**Actual Results**
 after 3 layers the layer begins to shift forwards

**Expected results**
 layers should not shift

**Additional Information**
 I have tested my printer on another slicer(S3D) with zero issues. This is occurring when printing via usb and SD card. This occurred after installing my tmc2130 drivers. I am using an mks gen L board. I am not sure if Cura doesn't agree with the drivers. Please contact me if there are further questions. I have analyzed the gcode and there is no indication of issues.

smartavionics, 2018-06-19:
Could you please make the gcode available, thanks.

AbeFM, 2018-06-19:
Are you controlling accelerations in the slicing? I bet you're running the y-axis too hard and causing skips. Try turning down accel and/or speed and/or jerk (or just turn off control in CURA) and see if it happens again.

dventu, 2018-06-22:
Thanks this can be marked as Solved. It was my cura acceleration and jerk setting causing the issue. I set the values the same as my marlin config and this solved it. I will slowly raise them to hope and see if I can speed up my prints.

closed 2018-07-10 by ianpaschal