Ultimaker/Cura #3956: Can someone tell me why Version 15 of Cura prints way better than 3.3 with the same settings

Ninjarev, 2018-06-19:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ninjarev, 2018-06-19:
I have a Maker select V2 and I switched to 3.3 cura from 15 and my prints have went way down hill 
I checked both programs for settings differences and they are the same but the 3.3 the prints are not near as good..   I can upload pics too

smartavionics, 2018-06-19:
What would be useful is if you provided the gcode for the same part sliced using similar settings with both of the slicers. Then I could compare them and maybe come to some conclusion. As it is, your report is worthless.

Ninjarev, 2018-06-19:
And here are the pics
[image]
[image]
[image]
[image]
[image]
[image]

smartavionics, 2018-06-20:
Thanks for the gcode - I've had a quick look and obvious differences are that one has the inner and outer wall z-seams aligned and the other doesn't so to get 3.3 to align the z-seams turn on the wall printing order optimization. Also, I notice that you have changed to use relative extrusion. I know nothing about your printer but I am wondering if the extruder resolution (steps/mm) is too low for successful use of relative extrusion. You may find that using absolute extrusion produces a better quality finish. If I see anything else worth mentioning I will come back.

Ninjarev, 2018-06-21:
WOW great thanks for looking that over  Ill try what you said and see if that helps it..

JIM

ianpaschal, 2018-07-10:
Almost 3 weeks, considering this question answered. Thanks Mr. Avionics.

closed 2018-07-10 by ianpaschal