Ultimaker/Cura #3957: Per-filament first-layer hotend/print temperature

VanessaE, 2018-06-19:
Different filaments don't just need different print temperatures, they need different first-layer temperatures as well.  Just setting a global first-layer temperature via the chosen quality profile is not good enough, as it would require a quality profile for each filament (and perhaps several per, if one has multiple general quality settings they might otherwise wish to use).

Getting the first-layer temp "close enough" might be okay when printing on plain glass or on glue-stick- or hairspray-covered glass, but one especially needs to be able to fine-tune that first layer temperature properly per-filament if the bed has something a little more advanced on it like Printbite, PEI, BuildTak, etc.  What works for one filament may cause too much adhesion for another, or perhaps it might not stick at all.

Compare with Slic3r (it also allows a first-layer bed temperature; I have no use for that, but I know others do).

(Moved from CuraEngine)

diegopradogesto, 2018-06-20:
We incorporated the PR #2333 before 3.3. That allows you to include any cura setting in your material profiles, so you can add the print temperature for the initial layer or some other settings that you think are important.

VanessaE, 2018-06-20:
With all the 47 bazillion files Cura creates.... seriously, there's a dozen files for every one legit profile, or so it looks.  With the way the files are named, it's confusing for a seasoned geek like me, so it'll be nearly impossible for a common user to figure out just what file to edit, nevermind what to put in it or where in the file to edit.

I recently started clean -- again -- by wiping-out *everything* Cura-related except the AppImage.

I then started-up, created one printer at that time (custom), then I created one material profile from scratch, switched to Custom Mode (as oppose to Recommended Mode) in the settings sidebar, selected one of the defaults, altered a few settings, and created a new profile from the changes (so, I did not use any "duplicate" options at any point).

I tested and tweaked and did all kinds of acrobatics trying to fine tune that one profile, eventually creating two more based on it (mainly because I was trying different speeds).

As I drew closer to the desired result, I eventually reduced it to just one quality profile.  I'm still working on it, but I figure once I get a good result on the one material, I'll create a second material, make further refinements to my one quality profile so that I get good results from both filaments, and then use the one quality as a baseline to create others.

As it stands right now, I have exactly one printer profile (containing exactly one extruder), one material profile, and one quality profile.

So remembering what I saw when I did a `find` earlier (to hunt down all of Cura's data so I could blow it away), I looked at just my configs:
 
```
vanessa@rainbird:~$ cd .local/share/cura/3.4/
vanessa@rainbird:~/.local/share/cura/3.4$ ls
cura.log	    machine_instances  quality		   themes
definition_changes  materials	       quality_changes	   user
definitions	    packages.json      scripts		   variants
extruders	    plugins	       setting_visibility
vanessa@rainbird:~/.local/share/cura/3.4$ ls definitions/
[this gave me no results]
vanessa@rainbird:~/.local/share/cura/3.4$ ls definition_changes/
custom_extruder_1+%232_settings.inst.cfg
custom_extruder_2+%232_settings.inst.cfg
custom_extruder_3+%232_settings.inst.cfg
custom_extruder_4+%232_settings.inst.cfg
custom_extruder_5+%232_settings.inst.cfg
custom_extruder_6+%232_settings.inst.cfg
custom_extruder_7+%232_settings.inst.cfg
custom_extruder_8+%232_settings.inst.cfg
Prusa+i3_settings.inst.cfg
vanessa@rainbird:~/.local/share/cura/3.4$ ls quality
[also no results]
vanessa@rainbird:~/.local/share/cura/3.4$ ls quality_changes/
custom_extruder_1_%232_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
custom_extruder_2_%232_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
custom_extruder_3_%232_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
custom_extruder_4_%232_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
custom_extruder_5_%232_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
custom_extruder_6_%232_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
custom_extruder_7_%232_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
custom_extruder_8_%232_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
custom_pla_norm%3A_15_mms%2C_0.4_x_0.08_mm_bead.inst.cfg
vanessa@rainbird:~/.local/share/cura/3.4$ ls materials/
custom_material+%232.xml.fdm_material
```

Um...wut?

I see what might be my actual printer and quality profiles, but while the one listed material could be mine, it contains enough stuff that has no relation to my setup, that I can't be sure (like references to a printer I don't have, with beds and nozzle options I didn't put in there).  There's 178 lines of XML for just a dozen actual settings that I'm given any control over in the UI.

That's not the worst of it.  When I went looking before, at a time when I had several quality profiles and several materials defined, it became ten times worse.

I'm sorry @diegopradogesto, but frankly, this config setup is an absolute disaster.  It may be computer-readable, but you can't expect a common end user to dive into this.

Plus....why are my printer and quality profiles filed under `*_changes`, and not in `printer/` and `quality/`? Also, what are all these custom extruders?

Ghostkeeper, 2018-07-31:
> in which we will probably include the possibility to add cura settings in the material profile.

I don't think this actually made it into the design, now. Better not keep your hopes up.

> like references to a printer I don't have, with beds and nozzle options I didn't put in there

Cura doesn't create the `Prusa+i3_settings.inst.cfg` file unless you added it. Maybe you deleted your configuration folder while Cura was still running?

> There's 178 lines of XML for just a dozen actual settings that I'm given any control over in the UI.

This is because people expect an empty material profile to behave the same as PLA. We copy the PLA profile if you create a new material.

> Plus....why are my printer and quality profiles filed under `*_changes`, and not in `printer/` and `quality/`?

Because technically you didn't define a new printer type but used an existing printer type (Custom FDM Printer) and made changes to it. Also, you didn't technically create new quality profiles but took the existing quality profiles and made changes to it. For the quality profiles we wanted to make this distinction because we want you to be able to reset your settings even if you saved them to a quality profile.

> Also, what are all these custom extruders?

Every extruder needs an extruder definition which sets certain things like which T-value needs to be output to the extruder switch command (`T0`, `T1`, etc) and what the nozzle offset is if there are multiple nozzles, and so on. We've pre-defined 8 of these extruders so you can have up to 8 extruders in Custom FDM Printer. Then we hide 7 of these by default.

NicholasTracy, 2018-09-07:
Shouldn't creating a new profile, new material, or custom printer start a blank clean slate? There seems to be many settings that are copied over that are not available in the GUI. And it seems intuitive that you would want to start with zero configuration and build from scratch rather than have an already filled template that you have to modify. Otherwise the 'create new' button would say 'duplicate'.

nallath, 2018-09-07:
A custom printer can't ever have a blank slate. Well... It could. But that would mean that a user is forced to enter 500+ setting values.

NicholasTracy, 2018-09-07:
500+? I'm not sure I understand why a printer would need 500+ setting values, Important ones should be a part of the wizard when creating a new printer and there seem to be a bunch missing from the GUI. The plug-in "Printer Settings" brings a few of these settings out in the GUI to the Quality Profile but these should be part of the wizard for custom printers. I guess my point is, Don't assume settings of a user's printer and not allow them to change those values.

nallath, 2018-09-10:
My point is that there is no such thing as a blank slate. Every setting *must* have a value. So if a user didn't set it, it uses the default (So there are always values that are "copied" one way or another). 
We understand that there are tons of differences between printers, but we can't put in the insane amount of effort it would take to build something like that. Especially not since almost all of us are paid by Ultimaker (so our main focus is making it better for UM's).

NicholasTracy, 2018-09-10:
Let me put it another way, there are 3 other slicers on the market that do exactly what i described. So explain to me again how there is no such thing as a blank slate?

fieldOfView, 2018-09-10:
Let's take an example: print speed. What is no speed?

NicholasTracy, 2018-09-11:
> Let's take an example: print speed. What is no speed?

You don't just start printing with a blank profile. you would have to go through a wizard (or even just a UI with the options) to set up your custom printer just like you do in any other slicer. The only way a profile should have default settings is if that profile is a preconfigured printer profile like for the UM2 or others. Custom Profiles shouldn't be duplicates of a known printer Unless you are willing to give the user full control of all parameters copied. Adding a custom printer should be "user defined". Obviously, you aren't going to just click "add custom printer" and start printing. If your printer is not listed in the printer profiles and you need to create your own you are going to want to set it up specifically for that printer. As of now, you have something that kind of does that but only gives the user options for a few things, there is a lot of configuration items that aren't available to the user that should be made available to anyone needing a custom profile.

I don't know the first thing about programming UI in python with QT and have no desire to set up the toolchain needed to develop for Cura, Or I would do it myself as a proof of concept.

nallath, 2018-09-11:
Those other slicers don't give you a blank slate. They also set certain defaults as a base and then ask you to change it (much in the same way as Cura does it). Perhaps they do a better job of convincing you that this is not happening in the background.

So no. There are no other slicers that do this. The main problem is and remains as fieldOfView puts it; What does no speed mean? What does no Jerk mean? What does no layer height mean? What does no infill pattern mean? Those are just 4 of the 500 settings that the slicer needs in order to generate a g-code. 

Since we assume that no-one wants to actually do that, we provide people a subsection of the settings. We might be a bit too aggressive in that, so that is something we can look at, but the fundamental problem isn't something that is as easy as you believe it to be. 

All that being said; Creating custom printers is a bit of an afterthought. You have much more control if you create the profiles in the "right" way; by adding quality profiles to cura without using the GUI. This isn't the most user-friendly way of doing it, but there are relatively few people doing it and it would take a huge amount of effort from our side to build. This brings us back to; People are free to build it, but since most developers are paid for by Ultimaker, we're not going to put the huge amount of effort in it to build it. We're busy enough as is with our own printer

NicholasTracy, 2018-09-12:
Then you leave out a huge market portion of users. I'm sure there are more people using Cura for a third party printer then there are people with actually Ultimaker machines. I for one prefer to build my own machines so that I know how to fix them and upgrade them as needed. That's not to say that I don't understand the situation at hand. But at the same time, having a Github repository and making the software open source means that feature requests are going to come up. I notice that no one is really on top of the issue in this repository or adding labels to things like actual bug reports or feature requests.

I didn't mean to derail the OP.  To get back on topic, Not having those options easy to change within the GUI is lazy. So is telling people to dig through XML style files in hard to navigate directory trees to make a "proper" profile.

fieldOfView, 2018-09-12:
> Then you leave out a huge market portion of users. I'm sure there are more people using Cura for a third party printer then there are people with actually Ultimaker machines

Realize that this market has no bottomline for Ultimaker. Ultimately, Ultimaker develops Cura for Ultimaker printers. There's a small group of developers that contribute code that works for their printers and every so often also benefits Ultimaker printers. But most of the work, especially on the front end, is done by Ultimaker-paid developers.

nallath, 2018-09-12:
It is lazy. You're completely right. But it's laziness out of necessity, not because of unwillingness to do the work.

NicholasTracy, 2018-09-12:
Off topic again, although still relevant to the point:

I understand your position, this is more directed to your bosses and team leaders than it is to you personally. I will be sending an e-mail out to plead my case for third-party hardware users and the effect contributing to the open source community has on the companies bottom line. In a nutshell, I believe you when you say you aren't paid to do that, but it might be within Ultimaker's best interest in the long run to pay you to put some eyes on that area specifically. Usually these things pay back eventually in the open source communities in one way or another.

Ghostkeeper, 2018-11-13:
We won't be making an interface to do this any time soon. If you'd like to start completely from scratch without any primers, here's a file that you can edit: http://dulek.net/work/empty_definition.def.json

There are 719 properties you have to fill in before it becomes usable though. I wouldn't advise starting in this way.

The original issue in this thread is one of the long-standing desires for some people: Being able to modify every setting in the lower base layers of Cura's profile system. Currently you're only able to do that for the top 2 layers: Unsaved settings and custom profiles. So far our argument has been that this is enough; people who make their own material profiles and re-use those for different quality levels will also be able to edit the XML files; after all, they must edit the config files to be able to do so.

I'd say that this still holds. Yes, some people would like to have a more powerful system, but we see no way to do that without making the interface more complex.

Enverex, 2019-04-03:
To get back on topic: As first layer print temp and first layer bed temp are both filament specific settings and print temp and bed temp are both listed as settings in the interface for a filament, would it be possible to add just those two explicit settings to the config page for a filament?

[image]

Otherwise you have to remember to set them every time you change filaments.

Ghostkeeper, 2019-04-04:
Print temperature and first layer bed temperature are not filament specific settings per sé. You'll need different temperatures to print at different speeds so they are more quality-specific settings.

I think eventually we'll want to list any setting there. The current list is there because of a limitation that we had up until firmware version 4.2 that caused the printer to ignore the material files if anything unexpected was stored in them, even if it was in a different namespace.

VanessaE, 2020-03-04:
Sorry to necro-post, but I decided to give Cura another go, and this issue came up.  Guess it's still needed, especially since the "offset" plugin I mentioned previously doesn't work on Cura 4.5.0.

Ghostkeeper, 2020-03-09:
You can give this a try with the Material Settings plug-in from the Marketplace. That should allow you to change this setting from the Cura GUI as well.

closed 2020-06-19 by nallath