Ultimaker/Cura #3959: Mesh Type Modify settings Overrides Extruder Choice

AbeFM, 2018-06-20:
As per: https://github.com/Ultimaker/CuraEngine/issues/798

Model printed in one material (extruder #1), when using a modify-settings mesh, will print with the material of the modifying mesh's extruder selection instead of the original object.

**Application Version**
3.4b

**Platform**
win10

**Steps to Reproduce**
Load a model. Add a support blocker. Press "move", select blocker. Resize, reposition.
Select 'per model settings', choose mesh type "modify settings".
Selected "compensate...overlaps" (doubt this matters)

**Actual Results**

[image]

**Expected results**
Object should continue to print with its own extruder. If the mesh is to modify this, it should be a selection under modify settings.

**Additional Information**
You're wasting my glue, son!

fieldOfView, 2018-06-20:
I would say this is expected behavior. If anything it should be impossible to change the mesh type of a support blocker.

diegopradogesto, 2018-06-21:
As I commented in the other thread, this is the expected behavior. I get your point that maybe the "support blocker" doesn't make really sense to assign it to a extruder, but since it's never printed, I don't see why we should not assign one.
When you load an stl and you choose "modify the settings" using "per object settings" it is assigned automatically (and always) to the default extruder, the same way when you load the model you want to print is assigned to the default extruder.

AbeFM, 2018-06-22:
The other case is, that you chose your object to have the extruder you would like, then you add this modifier, and it doesn't default to where you were.

It would be like if the modifier set the outer walls always to 1, or top pattern always to ZigZag. Great if that happens to be what you want instead of what you had selected, but I feel making changes that are non-explicit confuses people.

Ghostkeeper, 2018-08-06:
So how would you indicate that part of your mesh needs to be printed with a different extruder, then?

AbeFM, 2018-08-08:
A pull down menu that says which extruder to print with, 
[image]
without editing it too heavily, one can imagine that it says "Print Modified Objects With:" or something more clever. :-)

It's being broken off so it seems possible/easy and something I would use. Something similar could also be used to change supports from inside and under the object, coarsely.

fieldOfView, 2018-08-08:
Once thing that seems unrelated, but would actually make the behavior a lot more predictable, is that currently the geometry that is added by the Support Blocker tool can not be "shaded" correctly; it is always rendered grey. If you change the Mesh Type to "Print as Support", it should be rendered as support with a striped pattern. And if you change it to a "Modify settings", it should _render in the color associated with the extruder_ associated with the geometry.

The shading currently does not work because it is created without per-vertex normals. I will create a PR to fix this issue with the Support Blocker.

Ghostkeeper, 2018-10-15:
The pull down menu is implemented in the form of a set of buttons on the left. I think that sufficiently implements what you need while maintaining the power to print part of your mesh with a different extruder.

closed 2018-10-15 by Ghostkeeper

AbeFM, 2018-10-15:
I guess the logical thing to do here is to set the modifier to have the same extruder as the object it was created from - when you click on an object to add a blocker, set its extruder to match.
.
I agree the buttons let you change it. But just adding a modifier should modify as little as possible until it is set. Just changing the default would be enough.

Ghostkeeper, 2018-12-21:
This doesn't apply to support blockers though, they block the support regardless of which extruder they have.