Ultimaker/Cura #3960: [3.3.1] No Anycubic Kossel Linear Plus Printer Profile

CJ-Jacob, 2018-06-20:
**Application Version**
 3.3.1

**Platform**
 Windows

**Printer**
 N/A

**Steps to Reproduce**
 N/A

**Actual Results**
 N/A

**Expected results**
 N/A

**Additional Information**
 There are 3rd-party profiles available for the Anycubic Kossel Linear Plus printer, but these profiles are buggy and require much tweaking to have a satisfactory print result. This is an inquiry for a proper and official Cura printer profile for the Anycubic Kossel Linear Plus printer that optimizes print settings for print quality without dozens of test prints and wasted filament.

Appesteijn, 2018-06-22:
We don't have this printer at our office, so we can't make a good profile for it. If somebody out there has improvements, we are more that happy to add it.

ianpaschal, 2018-06-28:
That answers the question ^

closed 2018-06-28 by ianpaschal

MHaendel, 2018-07-03:
Could you share the one that you are running right now?

bagobor, 2019-09-26:
@Appesteijn good working profiles are here:  https://www.thingiverse.com/thing:2367365
It works like a charm in Cura 4.2.1

It will be awesome to have it build in. 
Thanks!