Ultimaker/Cura #3962: [3.3.1] Redundand M109 SXXX in gcode with dual extruder when setting constant temperatures

MaximilianBuegler, 2018-06-22:
**Application Version**
3.3.1

**Platform**
Arch Linux

**Printer**
BIBO2 Touch

**Steps to Reproduce**
Slice any model with dual nozzle. Such as supports printed with second nozzle. Set all temperatures for each nozzle to be equal (Printing temperature, initial, final, and **standby temperature**)

**Actual Results**
Every layer starts with M109 SXXX, causing the printer to wait a few seconds, costing precious time, while all XXX values stay constant for each nozzle. (Nozzle temperatures may still differ)

**Expected results**
Having only one M109 SXXX for each nozzle in the beginning (possibly another one after first layer, if first layer temp differs) to set both nozzle temperatures and wait for them to be heated.

**Additional Information**
A checkbox "keep nozzle at constant temperature during entire print" for each nozzle might be a nice thing.

MaximilianBuegler, 2018-06-22:
For others having the same issue and linux (/mac), I use 

`awk 'NR<=20{print}NR>20{sub("M109 ","");print} filename.gcode > newfilename.gcode'`

to fix the gcode files by removing all M109s after line 20. The 20s might need to be changed, depending on the size of your initial gcode.

AbeFM, 2018-06-22:
My whole first layer prints at my "initial" temp - the extruder is already warmed up, but cooled back down for this single material print.

`;FLAVOR:Marlin
;TIME:481
;Filament used: 0m, 0.760482m
;Layer height: 0.26
;Generated with Cura_SteamEngine 3.4.0-BETA
T1
M190 S62
M104 T1 S220
M109 T1 S220
M82 ;absolute extrusion mode
G28 W ; home all axes without mesh bed leveling
G1 X25 Y5;
T1; Switch Ext
G92 E0 ; reset extrusion distance
G1 X80.0 E12.5  F1000.0 ; priming
G92 E0; Zero Ext 1
G1 E-2.000 F2700; Retract
T0; Switch Ext
G92 E0 ; reset extrusion distance
G1 X120.0 E9.0  F1000.0 ; priming
G92 E0; Zero Ext0
G1 E-2.6000 F2700; Retract
G92 E0
G1 F3000 E-2.6
;LAYER_COUNT:2
;LAYER:0
G1 F2700 E-10
G92 E0
G1 X110 Y130
T1
G92 E0
M109 S216 *********************************************************
M107
G0 F3600 X258.013 Y26.229 Z0.3
;TYPE:SKIRT
G1 F1800 X258.053 Y25.852 E0.01986`

I could see how you might end up there, but the temp should be set to the full temp or "initial layer temp" before printing.

AbeFM, 2018-06-28:
As a workaround, it's ok. As a way the software should function, it leads to making a change to fake the system into working, then forgetting your off-label settings and ruining future prints until you remember, find, and revert the settings.

I would rather see the software do what it says on the box, less time spent on phone and internet support. :-)

MaximilianBuegler, 2018-07-20:
I don't quite understand your objection there, as this setting would only apply if standby temperature equals printing temperature. Hence only makes sense if pretty much every layer uses both extruders.

Well, anyways, I helped myself with a script replacing all M109s by M104s after initial heating:

https://github.com/MaximilianBuegler/CuraPostProcessingPlugins/blob/master/scripts/M109TOM104Fix.py

AbeFM, 2018-07-20:
I *do* use both materials on almost every layer.

I didn't think it was unusual to want to use the capabilities of the printer arbitrarily. :-P

If your script is working, then perhaps it's a good pointer for a way to fix the issue more broadly.

Ghostkeeper, 2018-08-13:
Fixed here: https://github.com/Ultimaker/CuraEngine/commit/fa7e4727dc70c7740a91bb52693243f7ca652460

closed 2018-08-13 by Ghostkeeper