Ultimaker/Cura #3965: [3.3.1]  Blank window on my 2nd screen

kas2, 2018-06-23:
**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
USB printing

**Steps to Reproduce**
Start V3.3.1 on main screen and move it to second screen  (so far so good)
Quit Application and restart

**Actual Results**
Blank window on second screen

**Additional Information**
Workaround:
Open C:\Users\...\AppData\Roaming\cura\3.3\cura.cfg
and reset "window_left" to zero:  window_left = 0

Thanks for this great software

kas2, 2018-06-23:
EDIT
An easier workaround:
Move the "blank window" from 2nd screen to main screen
Close application and restart

alekseisasin, 2018-06-29:
@kas2 , I tested on Windows 10, and Linux, for it worked fine. It might be related to your video card, drivers or etc. Thank you that you have found a solution by yourself.

closed 2018-06-29 by alekseisasin