Ultimaker/Cura #3966: Cura 3.3.1 broke saved Materials profiles

RolfTO, 2018-06-23:
**Application Version**
3.3.1 

**Platform**
Mac OS 10.13.5 

**Printer**
Ultimaker 2+ 

**Steps to Reproduce**
Try to load Material profile exported from the previous version of Cura 

**Actual Results**
Reports successful import but profile not added, even after program restart. Specifically my profiles for PET-G from FormFutura (ie; HD-Glass or Carbon Fil) 

**Expected results**
Material profile & setting should load and be available in dropdown. / Cura will no longer let me even specify PET-G as a material. 

**Additional Information**
I use FormFutura products for a lot of my projects with great success. Why on earth are their fantastic materials not part of the Cura availability list? I would like to think that an ostensibly "open-source" project like Cura would not be prone to exclusions of non-Ultimaker products. That being said, I can still find workarounds, like specifying my PET-G products as CPE+ and tuning from that base. That flavour of gcode seems to work fine. However it would be nice to have optimized profiles for a major vendor like FormFutura.  If you need my settings to create profiles, I'd be happy to contribute.  

Thanks, Rolf Berger / Toronto / Canada

alekseisasin, 2018-06-28:
@RolfTO 
To realize what went wrong we need your Material profile, then we will try to figure out what went wrong. Also could you please share your log file, from there we might get additional information.
You are right the PET-G is similar to CPE+, unfortunately I do not information about adding PET-G to Cura List

ianpaschal, 2018-08-01:
Haven't received further info like the mystery material profile for 1 month+

closed 2018-08-01 by ianpaschal

RolfTO, 2018-08-01:
Very sorry Ian.

I totally forgot about this. I’ll did it out when I’m home this weekend.

Rolf

ianpaschal, 2018-08-01:
OK. I'll reopen when you get around to it.

RolfTO, 2018-08-04:
Thanks for your patience Ian,

I haven’t programmed since the Fortran/Cobol/Basic days and haven’t searched for a log file in years. (Nice of Apple to hide the damn User/Library by default in High Sierra…) Took me a while and some googling to find the log file, but here it is and my saved materials too. I notice that there are several Carbon-Fil ones with greater degrees of xml in the file name, probably as I keep recreating them with upgrades.

As you can tell by the log I’m a sporadic hobby user; Only using Cura and my Ultimaker when I have some idea or practical need. Most times I boot up Cura, I have to update, and lose my materials in the process. I print a lot in PET-G, mostly HD-Glass and Carbon-Fil, with which I’ve had great success.  I have written versions of my profiles now so I can recreate them whenever I use those materials, but it’s always puzzled me why Form-Futura’s materials weren’t in the supported list. They’re awesome in my opinion.

Oh well, hopefully you can glean some insight from these files. Thanks for all your hard work keeping Cura solid. I understand the commitment and offer sincere kudos to those who toil in the shadows.

Cheers,

Rolf