Ultimaker/Cura #3967: [3.3.1] - Communication Settings not available - Can't connect to Printer

auBusinessDaD, 2018-06-24:
Ultimaker Cura 3.3.1 on Windows 10

I am using a Balco 3D (which is a clone of the Cocoon Create and Wanhoa I3) and am trying to add the printer to "Monitor" which I had in 3.2.1 using the instructions i used previously (https://winplus.zendesk.com/hc/en-us/articles/207788606-3D-Printing-Via-USB-Port) which I know are outdated but did allow me to adjust the COM port in 3.2.1.

I cannot find the settings to adjust the desired COM port for cura to look on to find my printer and as such Cura 3.3.1 cannot find my printer so that I can send and monitor my prints.

alekseisasin, 2018-06-28:
Hi @auBusinessDaD , Can you explain what do you mean under "add to the printer Monitor"
The printer should be found automatically, if it does not happen then might be it is security issue or etc. To understand it better please upload your log file then I will check what happens.

fieldOfView, 2018-06-28:
> I know are outdated but did allow me to adjust the COM port in 3.2.1.

Cura 3.2.1 had no way to set the COM port either.

alekseisasin, 2018-07-02:
@auBusinessDaD, It is strange, If you say that printer has worked in Cura 3.2.1 version. Could you please test again all three Cura versions and confirm that it still works in 3.2.1 and not in 3.3 and latest 3.4.

tommie, 2018-07-06:
I have the same problem with an Ender 2.

3.2.1 runs baudrate detection and eventually finds the printer, but 3.3 and 3.4 fail. I see some `PermissionError` in the logs, but not consistently. Mostly it just says "autodetecting baudrate" for a bunch of them, but nothing more happens in the logs.

I just today downgraded back to 3.2.1, and my printer works again.

ianpaschal, 2018-07-18:
Please share your logs.

tommie, 2018-08-18:
Sorry, no logs yet.

Observing my printer's display (Marlin), I see weird characters showing up in the status line, when on 3.4.1.

What's interesting is that my Marlin reboots when plugging in 3.2.1.

So I think what happens is that the baudrate detection causes random data to be sent, and Marlin either crashes hard enough to reboot (3.2.1), or it just locks up in the command parser (3.4.1). This is Marlin 1.1.x.

There were some changes to the baudrate detection code after 3.2, and maybe it was enough to change behavior. Is there any way to disable the autodetection?

diegopradogesto, 2019-09-11:
Still waiting for some logs... I'm going to close this ticket. Please feel free to reopen if it's still relevant.

closed 2019-09-11 by diegopradogesto