Ultimaker/Cura #3968: Reduce "empty" or "needless travel moves

Taiko1976, 2018-06-24:
**Application Version**
3.3.1

**Platform**
Windows 7

**Printer**
Selfmade RepRap

**Steps to Reproduce**
See attached G-Code file

**Actual Results**
e.g. starting from layer 26 the print head moves needlessly back an forth when printing the rings around the holes.

**Expected results**
no empty or needless moves to reduce nozzle drainage and reduced material when resuming print leading to holes in the print or thin walls.

**Additional Information**

[CFDMP_Coldend V2 - Housing.zip](https://github.com/Ultimaker/Cura/files/2130651/CFDMP_Coldend.V2.-.Housing.zip)

smartavionics, 2018-06-24:
Cura 3.3 has a regression that causes excessive travels. This is fixed in 3.4 so please try the beta and let us know if the problem is still there. Thanks.

julianjm, 2018-06-24:
@smartavionics Just tested the 3.4 beta and the long travel issue is fixed for me. Thanks.

Taiko1976, 2018-06-26:
Hi, looks like the 3.4 beta does not have that problem, or at least it is not as pronounced as in 3.3.1. For the moment I can see a big improvement, but will continue to monitor this problem. For the time being this issue should be closed. I'll open a new issue, if I'll encounter any problems with a specific part or some.

Thank you for your assistance... and keep up the good work. Thanks.

smartavionics, 2018-06-26:
Understood, thanks for the report. Please close this issue (I cannot do that).

closed 2018-06-26 by ChrisTerBeke