Ultimaker/Cura #3970: rotate object numerical

herrdeh, 2018-06-24:
**Application Version**
3.2.1 PPA
**Platform**
ubuntu 18.04

**Additional Information**
Rotating an object with the mouse is a pain. I'd really appreciate a numerical rotate control - like it is in existence at "move" and "scale". 

Thanks!

ChrisTerBeke, 2018-06-24:
Duplicate of https://github.com/Ultimaker/Cura/issues/2779.

closed 2018-06-24 by ChrisTerBeke

ChrisTerBeke, 2018-06-25:
The original ticket actually described why this is hard for rotation (the problem originates from math issues surrounding 3D transformations). Please add your comments to the original ticket if you think they add something new (as this one is closed).

npm12017, 2019-06-26:
the real question to be asked is, Why are you even considering 3d printing, if you can't figure out how to rotate an object by typing three numbers for each axis to have the object rotated by i.e. trial and error. 

When i used Simplified I changed the values accordingly until i was happy with the rotation of the object. In Cura, it is almost hair splitting just to do the same thing.

Blender does this well in three different ways:
By pressing the R button followed by the number of degrees be it - or + number. Freeroam rotate by double tapping the R button. Typing in the numbers in three different fields like in simplified3D.

The problem with the Gizmo, when I click on the axis I want to rotate by, its not a permanent click, and as soon as i turn the gizmo the thin line becomes unselected. 
***************************************************************************
I think a better approach to this UI experience is to go ahead and use whats there and add a double click  on axis to enter rotation/ double click on empty space to exit rotation experience
**********************************************************************************

It would be great if we could double click on the thin line, then highlights to rotate the mouse is then locked to it. Allowing the user to then double in empty space to exit the locked rotation mode.