Ultimaker/Cura #3971: keayboard menu controls do not work in Linux

herrdeh, 2018-06-24:
**Application Version**
3.2.1 PPA
**Platform**
ubuntu 18.04 GNOME

**Steps to Reproduce**
hit alt-d or alt-b or alt-a (I'm at the German version)
**Actual Results**
nothing
**Expected results**
The "Datei" / "Bearbeiten" / "Ansicht" menu should pop up.

alekseisasin, 2018-06-29:
@herrdeh , I think the issue is related to this discussion: 
https://github.com/Ultimaker/Cura/issues/2964#issuecomment-352139776

Ghostkeeper, 2018-08-10:
It's not the keyboard layout thing or specific to Linux. It's the language. Switching to German reproduces this problem for me too.

Ghostkeeper, 2018-08-10:
I get a warning in the log when these buttons are pressed:

    QQuickAction::event: Ambiguous shortcut overload: Alt+B

DDDirk, 2018-08-12:
I have a similar problem using Cura 3.0 AppImage (also on Linux).
Alt - F, E, V, X and H work (File, Edit, View, Extensions, Help), but nog Alt - S, L and R (Settings, Plugins and Preferences). I also get the 'Ambiguous shortcut overload' messages in the terminal for those shortcuts.

Also (possibly related?), moving between the menus with the arrow keys doesn't work everywhere. Using those I can get to Settings, with Alt-V and then right arrow. Another right arrow opens the Printer submenu (because the top item gets preselected), yet another closes the menus and and yet another one starts rotating the build plate. If I do Alt-X for Extensions and then right arrow repeatedly, nothing happens at all. Maybe this has something to do with that the top item (Changelog) does not get preselected? This only happens in Extensions. In all the other menus, the top item gets preselected (if not greyed out).
Going left (from Help) works ok all the way.

diegopradogesto, 2018-08-15:
For Devs.: CURA-5642

diegopradogesto, 2018-08-15:
It happens also in Windows and also in English.

Ghostkeeper, 2018-10-12:
We fixed this in version 3.5. Thanks for the report!

closed 2018-10-12 by Ghostkeeper