Ultimaker/Cura #3972: [3.4.0-BETA] Arrange All Models not working

upatamby, 2018-06-24:
**Application Version**
3.4.0-BETA

**Platform**
Windows 10 Pro

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Open Cura
Load test.stl
Right-click on the model and choose: Multiply Selected Model
Select Number of Copies: 2 and press OK
Select Edit, Arrange All Models

**Actual Results**
After Multiply Selected Models the added models are placed at a spot where they cannot be printed.
After Edit, Arrange All Models, the placement is NOT adjusted, however, I am able to manually adjust the placement so that all 3 of the models can be printed.

**Expected results**
After Multiply Selected Models and after Edit, Arrange All Models I expect that the models are placed on the right spot, i.e. a spot where they can be printed.

**Additional Information**
Everything worked fine in Cura 3.3.1
[test.zip](https://github.com/Ultimaker/Cura/files/2130791/test.zip)

ChrisTerBeke, 2018-06-25:
@jackha ping!

nallath, 2020-11-03:
This has been fixed for 4.8. We are now using a much better algorithm for this.

closed 2020-11-03 by nallath