Ultimaker/Cura #3974: [3.3.2] fontconfig error cura did not start

artbody, 2018-06-25:
Gentoo Linux

```
 ./Cura-3.3.1.AppImage
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number

```
Cura didn't start .

same Error with 3.2.1 but cura starts and does it's job quite well

smartavionics, 2018-06-25:
Works for me OK on Ubuntu with these messages:
````
QApplication: invalid style override passed, ignoring it.
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
````

ChrisTerBeke, 2018-06-25:
This has been reported several times before. These 'shady' versions of Linux sometimes give font issues, but we can't really debug why.

artbody, 2018-07-09:
so one step further
i switched off the 10-scale-bitmap-fonts.conf 
```
eselect fontconfig list
...
 [7]   10-scale-bitmap-fonts.conf *
... 
```
`eselect fontconfig disable 7`
But now both versions of cura are crashing WITHOUT any message !!!

So i think the font is not the problem of crashing

Ghostkeeper, 2018-08-17:
Maybe it's an idea to run Cura through GDB, see what exactly is causing the crash if it's not the fonts.

artbody, 2018-08-23:
this is the meassage with cura 3.4.1 which starts without crashing 
but only under root account
```
artbody ~ # /opt/cura/Cura-3.4.1.AppImage
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range

```

Ghostkeeper, 2018-10-25:
Let's assume this is fixed then in 3.4+.

jayeye, 2019-07-30:
The problem persists in Cura 4.2.0 under Ubuntu 18.04

smartavionics, 2019-07-30:
Hello @jayeye , please zip stdout.log and stderr.log which should be in ~/.local/share/cura and cura.log which should be in ~/.local/share/cura/4.2.0 and attach to this issue. Thanks.

jayeye, 2019-07-30:
as requested.
Lots and lots of warnings, by the way. 
[3974.tar.gz](https://github.com/Ultimaker/Cura/files/3448691/3974.tar.gz)

smartavionics, 2019-07-30:
Thanks. I didn't get the cura.log, do you have that also?

jayeye, 2019-07-30:
Ah, it was in the 4.2 directory. Here it is.
[cura.log.gz](https://github.com/Ultimaker/Cura/files/3448709/cura.log.gz)

smartavionics, 2019-07-30:
I'm confused, from cura.log it looks like you can slice OK. What exactly is  the problem you have right now?

smartavionics, 2019-07-30:
Completely dumb question but you have made the AppImage file executable, haven't you?

BredanDE, 2019-12-08:
Had the same issue, pretty spontaneous. This tip here [https://www.networkshinobi.com/cura-in-linux-fontconfig-error/] has helped (delete qtshadercache). The errors still appear but CURA starts with readable windows instead of white boxes.

closed 2020-06-19 by no-response[bot]