Ultimaker/Cura #3975: Active leveling - wrong print core loaded

welzig, 2018-06-25:
**Application Version**
<Cura V3.3.1>

**Platform**
<Windows 10>

**Printer**
<Ultimaker 3>

**Steps to Reproduce**
<Run active leveling>

**Actual Results**
<The printer start active leveling with the second printhead in active position instead of first one, which ends in a collision of the printhead and the upwards moving building plate because it's out of sensor range>

**Expected results**
<At the beginning of active leveling printer have to move second printhead in passive position to avoid collisions with the build plate before reaching the sensor detection level>

**Additional Information**
<Issue known since latest firmware update V4.3.3 V2018-05-29>

ChrisTerBeke, 2018-06-25:
This seems like a firmware issue. Please contact your original Ultimaker reseller for support on hardware/firmware (the know a lot more about this).

closed 2018-06-25 by ChrisTerBeke

welzig, 2018-06-26:
Sorry, but these confuse me. Is this not the service contact from ultimaker for all issues or only for Cura related topics?

 

Mit freundlichen Grüßen - best regards

 

 

Daniel Welzig

 

 

aSpect Systems GmbH 

Eisenbahnstr. 2

01097 Dresden        

Germany       

 

Tel: +49 351 899 67 55      

Fax: +49 351 899 67 84      

email: welzig@aspect-sys.com 

 

This message may contain confidential information. If you are not a designated recipient, you may not review, copy, distribute or disclose this message. Please notify the sender by reply e-mail and delete this message. Thank you.

fieldOfView, 2018-06-26:
> Is this not the service contact from ultimaker for all issues or only for Cura related topics?

No, it isn't. It is a hub for developers (from Ultimaker and 3rd parties) to discuss development of Cura, not a support forum. As indicated by @ChrisTerBeke, this seems to be a firmware issue, and the firmware is developed by a whole different set of developers, who don't frequent this issuequeue. The firmware is (at this time) not developed "in the open", so you were pointed to a better way to get your problem solved (through your reseller). Another way would be to post at the Ultimaker forum (https://community.ultimaker.com/forum/106-ultimaker-3d-printers/)