Ultimaker/Cura #3979: Save project causes reslice

BagelOrb, 2018-06-26:
**Application Version**
master, 3.4 beta, etc.

**Platform**
Linux mint 18.2 Cinnamon (64bit)

**Printer**
UM3, etc.

**Steps to Reproduce**
- Load (not too simple) model
- Let slice
- Save project

**Actual Results**
Model is resliced

**Expected results**
Model is not resliced

**Additional Information**
Such a reslice is very annoying, especially when you just sliced a big print (e.g. for the UM S5) and you want to save the project just before you send it to the printer.

Saving the project just before you send a print to the printer is very useful, because it gives you a steps-to-reproduce for whenever in the future you would have to redo the print.

smartavionics, 2018-06-26:
I griped about this some time back and I think it was @Ghostkeeper who said something like "we don't care and we're not going to fix it".

BagelOrb, 2018-06-26:
Alright. I'll close it then 

D':

Ghostkeeper, 2018-08-08:
Uh, I don't think I would've said that.

Perhaps you were confused by a similar issue: Saving a custom profile causes a reslice. We don't think that is as important because it doesn't occur very often.

Ghostkeeper, 2018-08-08:
I'm unable to reproduce this problem though. I've loaded various models, complex and simple, but it doesn't reslice.

BagelOrb, 2018-08-08:
Nope. Can't reproduce either.

:tada: :yum:

closed 2018-08-08 by BagelOrb