Ultimaker/Cura #3980: [3.3.1] Compatible Material Diameter not saved in printer settings

SnowZA, 2018-06-26:
**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
Custom FDM printer

**Steps to Reproduce**

1. Create a new printer based on the Custom FDM printer.
2. Adjust the compatible material diameter field on the Extruder tab of printer settings to be 1.75 instead of 2.85.
3. Close, and then re-open Cura.

**Actual Results**
The compatible material diameter field is reset back to 2.85.

**Expected results**
The compatible material diameter field should be what I set it to.

**Additional Information**
The chosen material diameter is not saved in any form in any of the printer or extruder profile files. After adjusting the width in the settings dialog, it displays as the set width when closing and re-opening the dialog, but the materials list shown is still wrong. Once Cura is closed and re-opened, the "material_diameter" {"default_value"} is loaded in again from the fdmprinter.def.json file.
There doesn't seem to be any material diameter settings saved either in the printer config, or the extruder config files that are created.
When I edited the "material_diameter" {"default_value"} in the json file and then opened Cura again, I then see the correct material list and the extruder tab has the 1.75 value. 

There is also a "material_diameter" {"default_value"} field in the fdmextruder.def.json file, but editing this doesn't seem to have any effect. The compatible material diameter field seems to be being overridden by the setting in the fdmprinter definition.

I realise that there are 1.75mm printers that I can use as a starting point for my printer, but having the compatible material diameter on the extruder tab implies that it should be possible to have multiple extruders with different compatible material diameters on one printer.  With the way it is currently working, however, that is not possible. Also, since that field is editable in the dialog, one would expect that to override whatever is in the default definition.

LipuFei, 2018-06-28:
Hi @SnowZA , this is probably a duplicate of #3378. We have made a fix in 3.4. Could you try 3.4?

LipuFei, 2018-06-28:
@SnowZA Ah, sorry, it's not in 3.4 but in master... So it will be fixed in the next release.