Ultimaker/Cura #3981: nozle size

louisthissen, 2018-06-26:
**Application Version**
3.4.0

**Platform**
macos 10.13.5

**Printer**
ANYCUBIC Kossel Linear Plus

**Steps to Reproduce**

I have an 0.4 mm nozle, but if i enter this size, after restarting the program the setting sets itself back to 0.5 mm. And by slicing my peaces, it is visible that the parameter is not right. but it stil works after all, it is just not perfect.

fieldOfView, 2018-06-26:
Try entering 0.401 as nozzle size as a workaround. Not perfect, I know, but a workaround that gives better results than using 0.5 with a 0.4 nozzle.

louisthissen, 2018-06-26:
seems to work, for a temporary (long term) solution it is a good idea. thank you ;-)

Appesteijn, 2018-06-28:
Looks like this bug was fixed in https://github.com/Ultimaker/Cura/issues/3378. We will investigate.

diegopradogesto, 2018-06-28:
With the issue report #3773 we created a ticket and it's in our board now. It is fixed apparently in master but it needs to be tested to confirm fully working.

closed 2018-06-28 by diegopradogesto