Ultimaker/Cura #3984: loading Monoprice Mini Delta profile crashes

tmbdev, 2018-06-27:
Cura version: 3.4.0

Go to https://www.mpminidelta.com/slicers/cura

Download the MPMD definition

Install in ~/.local/share/cura/3.4/definitions

Try to add MPMD printer

Result: Cura crashes

Expected behavior: if the profile is incompatible or contains errors, an informative error message would be nicer.

It would also be nice if a working MPMD profile could just be added to Cura by default.

ChrisTerBeke, 2018-06-27:
This definition is not in the official Cura build and not checked by an UM developer to validity, so it's hard to say anything about that. It's better to contact Monoprice about this so they can fix it.

closed 2018-06-27 by ChrisTerBeke

fieldOfView, 2018-06-27:
@ChrisTerBeke:
> if the profile is incompatible or contains errors, an informative error message would be nicer.

Cura should not crash, even when it is fed incorrect data

fieldOfView, 2018-06-27:
On a totally unrelated note, it is interesting how monoprice deeplinks to Cura builds on software.ultimaker.com, skipping the survey.

ChrisTerBeke, 2018-06-27:
Monoprice does a lot of interesting things :P

True that Cura shouldn't crash, but I think it gives the actual 'crash report' dialog asking you to reset your data instead of just closing?

thetrebor, 2018-06-30:
The MPMD profile typically includes this JSON in the printer definition which crashes Cura:
"layer_height": { "minimum_value": 0.05, "maximum_value": "machine_nozzle_size * 0.75", "maximum_value_warning": "machine_nozzle_size * 0.5 + 0.01",  "default_value": 0.10 , "value": 0.10 },

it doesn't crash if you remove everything except "default_value" from the object.

thetrebor, 2018-06-30:
Traceback (most recent call last):
  File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 898, in getQualityProfilesDropDownMenuModel
  File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/QualityProfilesDropDownMenuModel.py", line 47, in __init__
  File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/QualityProfilesDropDownMenuModel.py", line 70, in _update
  File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/QualityProfilesDropDownMenuModel.py", line 107, in _fetchLayerHeight
TypeError: float() argument must be a string or a number, not 'SettingFunction'

adiastra, 2018-07-23:
Do we have a fix for this (or a proper profile for MPDM/C3.4)?

thetrebor, 2018-07-23:
I believe the more correct bug report is being tracked in this ticket:
https://github.com/Ultimaker/Cura/issues/4005

adiastra, 2018-07-23:
Thanks, i see your note too. my MPMD is not here yet, when it gets here, what should I do for a printer profile?

adiastra, 2018-07-23:
nvm, ill just use an older version of cura I guess.

thetrebor, 2018-07-23:
you should be fine as long as you don't use the wiki version of the mpmd profile.  or if they've updated the profile.  it's not a problem with the printer, it's a problem with the combination of cura and a profile that makes use of all cura's features.

adiastra, 2018-07-23:
Do you have a good profile for 3.4 you are willing to share?

thetrebor, 2018-07-24:
Yes, i'll DM you a link tonight.

tmbdev, 2018-07-24:
This isn't a bug report about the MPMD not working, it's a bug report about how Cura deals with incorrect profiles. The current behavior (crashing) makes it hard to debug profiles.

#4005 addresses one particular problem, but in general. the code that loads profiles should be made robust against crashing by wrapping it in recovery code: try...except and try...finally, and it should give information about which part of the file it was processing before the error occurred.

thetrebor, 2018-07-24:
I don't disagree.  But this one was closed.  So i made a bug with distinct, precise, actionable data so the devs can pick it up without having to work too hard.  A generalized enhancement like fault-tolerant profile parsing is not something that will be grabbed by a dev readily.

luisriverag, 2019-09-23:
> Yes, i'll DM you a link tonight.

Would you be so kind as to PM me as well?

thetrebor, 2019-09-23:
regrettably after I damaged my MMPD I haven't maintained my profile for it.  Sorry!