Ultimaker/Cura #3985: print sequence should take into account extruder geometry

tmbdev, 2018-06-27:
As far as I can tell, when placing multiple objects and printing them "One at a time", the print sequence seems to be "order added" or something like that. But the optimal print sequence depends on extruder geometry. For example, for an Anycubic i3 Mega extruder, the fan is to the left of the nozzle, so printing objects right-to-left allows them to be spaced more closely.

Ideally, placing and print ordering for multiple objects would take into account extruder geometry. However, at least a preferred print order (specified perhaps as a vector) would already be quite helpful.

ChrisTerBeke, 2018-06-27:
Cura does this, but it might be that your machine configuration does not have the right geometry in it?

tmbdev, 2018-06-28:
I see no such configuration option either in the printer definition or in the extruder definition; where would that be found, for example for the Prusa mk2?

ChrisTerBeke, 2018-06-28:
It's not configurable in the UI, only in the source files of the printers.

UM3: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/ultimaker3.def.json#L42
Fallback value: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json#L414

The creators of the Prusa mk2 and Anycubic i3  did not set these values, so it uses the default fallback value.

closed 2018-06-28 by ChrisTerBeke

fieldOfView, 2018-06-28:
> It's not configurable in the UI,

Yes it is, that's what the Printhead Settings in Machine Settings control.

ChrisTerBeke, 2018-06-28:
@fieldOfView ah cool. I never dive into those :P