Ultimaker/Cura #3987: firmware update not working i was trying to upgrade my ender 3 frimware it was lagging out

chrisbeaudoin, 2018-06-27:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

alekseisasin, 2018-06-28:
@chrisbeaudoin , First of all could you please fill in the issue form correctly, How the "ender 3 firmware" is related to Cura?

chrisbeaudoin, 2018-06-28:
Sorry for not filling the form correctly ok im trying to update my ender 3 firmware with the cura software like the company was telling me to do when I go to cura I go to manage printer update firmware I choose the .hex file but the loading bar doesn’t move the printer company said to contact you guys thasnks for your time

alekseisasin, 2018-06-29:
@chrisbeaudoin, It is still hard to reproduce the issue without having your printer. What you can do, try to update firmware in different Cura version, like latest 3.4, or 3.3 or 3.2 etc. We want to be sure that it worked before.
I think the vendor of your printer also should have another way to update printer firmware.

fieldOfView, 2018-06-29:
Duplicate of #3731

closed 2018-07-09 by ChrisTerBeke