Ultimaker/Cura #3989: Machine Settings : bad option Origin at center

Tpelleterat, 2018-06-27:
Hello

If in machine setting I check **Origin at Center**, the print start in zero.
This is bad label.

Informations : 
 - Cura 3.4.0
 - Creality CR 10 mini
 - G-code RepRap

ianpaschal, 2018-06-28:
Please fill out issue template including attaching pictures or project files, or else this issue will be closed.

vincenzonardo, 2018-07-01:
Salve  ho acquistato una stampante DUPLICATOR I3 PLUS  ha funzionato bene.
Da questo progetto quando inizia   a stampare la testina non si sposta al centro del piano di stampa, ma rimane nella posizione iniziale  stampando 1/4 dell'oggetto.
in fase di settaggio il programma sistema in automatico l'oggetto al centro del piano di stampa. 
Hi I bought a DUPLICATOR I3 PLUS printer it worked fine.
From this project, when it starts to print, the head does not move to the center of the printing plate, but remains in the initial position by printing 1/4 of the object.
in the setting phase the program automatically arranges the object at the center of the printing plan.
[image]
[image]

ianpaschal, 2018-07-01:
@vincenzonardo In theory, please open a separate thread for your issue rather than taking over this one. In actuality though, I see you are using Wanhao's fork of Cura 2.x which we do not maintain or support.

fieldOfView, 2018-07-02:
*Turn off* the option "Origin at center" in Machine settings. This is only meant for delta-style printers.

ianpaschal, 2018-07-02:
@fieldOfView Yes, but the arguments that it's poorly labelled might be valid. It can be pretty confusing that Cura treats the center of the build plate as the origin when translating a model but the corner as the origin when printing.

I'm hoping for some clarification on

>the print start in zero.

though.

fieldOfView, 2018-07-02:
I created the options in Machine Settings to mirror the machine settings in legacy Cura. I think adding something to the tooltip is the best way to go forward. "This option is normally only applicable to delta-style printers".

ianpaschal, 2018-07-09:
Yes. We discussed this at Cookies&Cura and believe that any possible label could be confusing and so the better choice is to leave it as is, but integrate an explanation into Aleksei's Cura Guide plugin.

closed 2018-08-01 by ianpaschal