Ultimaker/Cura #3990: Zoom is limited for tall printer dimensions -- cannot zoom to fit

rtideas, 2018-06-27:
**Application Version**
3.4

**Platform**
MacOS 10.13.5

**Printer**
Custom printer: 500(x)x500(y)x1800mm(z)

**Steps to Reproduce**
Create custom printer with 500 x 500 x 1800 mm dimensions.
[image]

**Actual Results**
Does not show full volume.

**Expected results**
Should be able to zoom out to show full volume--often called zoom to fit.

**Additional Information**
Prototyping a tall printer for a project. Created a machine setup that is 500mm(x) x 500mm(y) x 1800mm(z). I appears you are limited to har far out you can zoom.

Would assume you could zoom to fit.

[cone.3mf.zip](https://github.com/Ultimaker/Cura/files/2142687/cone.3mf.zip)

Standard 3d view
[image]

3d view zoomed out to the max. Cannot see full object.
[image]

front view
[image]

front view zoom out to max.
[image]

front view panned all the way to the top. Can only see a partial section of the object--never in full.
[image]

top view panned -- Can see the object full, but not very useful.
[image]

ChrisTerBeke, 2018-06-28:
You can pan by holding shift and then drag, or drag using the middle mouse button.

fieldOfView, 2018-06-28:
OP obviously knows you can pan. Panning does not fix the problem.

ChrisTerBeke, 2018-06-28:
I don't see what would solve the problem then, other than having a monitor with a different aspect ratio... I can zoom out fine to show the whole build volume (even when it's tall).

fieldOfView, 2018-06-28:
I have an Ultimaker with Z Unlimited XXL which has a 1800 mm high build volume, and I can't get even close to seeing the whole buildvolume by zooming out and panning.

rtideas, 2018-06-28:
Yes,the issue in not panning. There is a finite zoom level. The maximum zoom out value should be constricted by the machine volume.  

I would assume there is a value setting the minimum zoom level or visual scaling. Possibly it is derived from the maximum zoom?  

Changing to a theoretically 2000mm cube results in same limited zoom out ability. 

Without this ability, the ‘big picture’ cannot be seen.

rtideas, 2018-06-28:
Would it have something to do with camera_distance?

https://github.com/Ultimaker/Cura/blob/master/resources/shaders/camera_distance.shader

Ghostkeeper, 2018-08-09:
No, it's limited by this variable: https://github.com/Ultimaker/Uranium/blob/2ffdea15427b10eb58237f1e1928bf2cfa6bc9ee/plugins/Tools/CameraTool/CameraTool.py#L28

Ghostkeeper, 2018-08-21:
It now sets the maximum zoom to some multiple of the length of the diagonal of the build volume.