Ultimaker/Cura #3991: Cura 3.4 crash when Add custom printer

gaerank, 2018-06-28:
**Application Version**
3.4

**Platform**
Fedora27

**Printer**
Custom printer

**Steps to Reproduce**
Add new printer type as custom printer. click "Add Printer" button then crash.
cura closed.

**Actual Results**
cura crash. end of program.

**Expected results**
process custom printer setting.

**Additional Information**

LipuFei, 2018-06-28:
Hi @gaerank, could you provide the log files? Thank you.

gaerank, 2018-07-02:
I found  " ~/.local/share/cura/stderr.log "
Is this right file??
[stderr.log](https://github.com/Ultimaker/Cura/files/2153536/stderr.log)

gaerank, 2018-07-02:
Hi, I've solve the problem.
I delete all from ~/.config/cura and ~/.local/share/cura.
and run again. it works fine.

Thank you.

closed 2018-07-02 by gaerank

Ghostkeeper, 2018-08-13:
Whatever it was, I believe this particular bug was fixed anyway when @diegopradogesto saw your bug report. Fix is here: 762a1b1bfdbb7fcb37bfa08013870793f5d20640