Ultimaker/Cura #3994: Cura 3.4 Pauses print after heating and again after bed leveling

leeb65, 2018-06-28:
**Application Version**
3.4.0

**Platform**
Windows 10, version 1803

**Printer**
CTC Prusa i3 Pro B DIY
Modified with Marlin 1.1.8 and 3D Touch bed levelling probe.

**Steps to Reproduce**
Load STL, Prepare and print

**Actual Results**
The printer heats up bed and nozzle, then on the printer screen it says press button to resume, I then have to press the button twice and the printer goes through the bed levelling and stops again with the message Click to Resume. I press the button and it prints as usual.

**Expected results**
The printer should heat up, level the bed and print without interaction.

**Additional Information**
I have run the same print through Octopi and it runs as expected, so it is not the gcode.

[PI3_Toilet_Seat_Bumper_v1-0.zip](https://github.com/Ultimaker/Cura/files/2144739/PI3_Toilet_Seat_Bumper_v1-0.zip)

[cura_Log.zip](https://github.com/Ultimaker/Cura/files/2144944/cura_Log.zip)

alekseisasin, 2018-06-28:
Hi @leeb65, Unfortunately we do not have CTC Prusa i3 Pro B DIY in our office and I do not know how to test. Can you run the gcode from USB stick? Also, did you have same issue in Cura 3.3.1?

leeb65, 2018-06-28:
Hi, No I did not have the problem until I updated to Cura 3.4.0 yesterday. I thought it might be my printer, so I connected it to Octopi and saved the gcode from Cura, copied it to Octopi server and set the print off from there. It ran from start to end without a problem.

alekseisasin, 2018-06-28:
I know we did fix for the auto bed leveling (https://github.com/Ultimaker/Cura/issues/3808), but this is a different issue. (probably if you know how to search in github they were related to ticket CURA-4844)

neowinx, 2018-06-28:
I have to report that I have this same issue with my Geeetech Prusa Pro W with BL Touch sensor.

After updating Cura to 3.4.0 it pauses at these same spots (after heating and leveling) displayin a "click to resume" message.

Would do some testing to see if this problem only occurs with marlin 1.1.8 with BLTouch.

**Application Version**
3.4.0

**Platform**
Windows 10, version 1803

**Printer**
Geeetech Prusa i3 Pro W
Modified with Marlin 1.1.8 and 3D Touch bed levelling probe.

leeb65, 2018-06-29:
The CTC is basically a Geeetech, I have the GT2560 Rev A+ and the similar settings in configuration.h

maikbooehme, 2018-07-02:
hi all, I have the same issue with an AM8 and Marlin 1.1.8 .... so thats not printer specific and started after the upgrade to CURA 3.4  what did you change in the code ?

ianpaschal, 2018-07-02:
We just discussed this as a team. Unfortunately we don't have these printers to test with so the only way we can make any fixes/improvements is to know exactly what part of the gcode needs to be changed.

maikbooehme, 2018-07-04:
So I checked it again and the issue is within the start Gcode of the printer. 
If the following is set, then the system is waiting for a click to resume printing before autobed leveling
M109 S{material_print_temperature} ;Wait for extruder to reach temp before proceeding

leeb65, 2018-07-04:
If it was the gcode, why did it work in every other version of cura and still work in Octopi?

reforket, 2018-07-18:
I have the same problem.
Printer: Flyingbear Tornado
Firmware: Marlin 1.1.8 with BLTouch

maciekbaczmanski, 2018-07-23:
Hi! I have the same issue :(
Printer: Tevo Tarantula with mks gen L board
Firmware: the newest Marlin firmware downloaded from [https://github.com/JimBrown/MarlinTarantula](here)

leeb65, 2018-07-23:
I have stepped back to 3.3.1 until this bug is fixed, if it does get fixed.

sevenmead, 2018-08-09:
Same for me when I upgraded to 3.4 on a rigidbot upgraded with arduino ramps. Was printing flawlessly and the upgrade cause massive issues. I'll downgrade.

CodeOptimist, 2018-08-09:
Kudos on the other improvements in 3.4. 👍😉 The pausing is causing me issues as well. My end g-code heats up the bed extra for a minute, I've found it makes a huge difference in releasing the print. But 3.4 broke it. :(

tre4b, 2018-08-09:
Okay.  I have been using CURA 3.4 for a while and have a Geeetech i3 running 1.1.9 now but 1.1.8 up to a while ago and I've not seen these issues. Until today.

Today I had my first go with something other than PLA (PETG) and used a higher nozzle and bed temperature.  I then got this message and had to press to start the print.  When this finishes (and it is currently printing very badly) I will try again with PLA and see if the message goes away.

tre4b, 2018-08-09:
And the result is.  The printer is happily printing PLA without me having to press a key.  There is some correlation between material or temperature going on here.

CodeOptimist, 2018-08-09:
For me it's `M190 S<bed temperature>`. @maikbooehme above mentioned the same but `M109` for extruder. Didn't pause before 3.4, just waited for the temperature as described.

leeb65, 2018-08-22:
It's quite obvious the problem is at the end of the heating of bed and hotend, also just before printing, a pause signal is sent to the printer. Remove the pause code and voila it should print without pausing. 

The other alternative is to use one of the other slicers with print software or Octopi. It cannot be difficult to compare the code from v3.3 with the code from 3.4 and see the differences, there I loads of software out there. I would not get away with releasing a piece of software at work that doesn't do it's job properly, I cannot say 'oh it won't work in Chrome so you can't use that.', we even have a hat for the 'it works on my machine'.

fieldOfView, 2018-08-22:
Hostility gets you nowhere here. None of the developers here owes you anything.

Ultimaker developers develop Cura for Ultimaker printers, because they are paid by Ultimaker. Other, mostly volunteer developers, like me, develop Cura because it helps our own situation, and because we derive some “fun” from it. Do you think your saying we are stupid or threatening to use other software contributes in any way to that “fun” or my ability to use Cura myself? Spoiler-alert: no, it doesn’t.

As for your suggestion to just look at what changed in USBPrinting between 3.3 and 3.4, that is exactly what I did above. No, there’s no pause being sent.

Cura does not work for you? Sorry to hear that, come talk to us in a civilized way. Can’t do that? Then PLEASE go use some other software. I owe you nothing, you owe me nothing.

leeb65, 2018-08-22:
I'm sorry, if making a suggestion to someone to use another slicer offends you? Is it not better than me swearing at you for it not being fixed? 

I have talked to you and others have, one person swears and you take offence to a suggestion.

CodeOptimist, 2018-08-22:
fieldOfView may have mistaken you for zinenu

ianpaschal, 2018-08-23:
^ I think so too, although @leeb65 you're polite but also extremely patronizing.

> It cannot be difficult to compare the code from v3.3 with the code from 3.4 and see the differences, there I loads of software out there.

See the differences, sure. Understand why exactly something changed, not easy. Also has nothing to do with what other software is out there.

> I would not get away with releasing a piece of software at work that doesn't do it's job properly

Good for you and the world thanks you for it. We also do extremely extensive testing on Cura bug bugs still arise. ;)

alekseisasin, 2018-08-23:
By the way. At the beginning of the conversation was mentioned that main difficult for Cura team is that we do not have the 3Dprinter where the issue has occurred (_Unfortunately we don't have these printers to test ..._). 
You can help us to find the problem if you would carefully identify the problem ( to be sure that it is not any mechanical issue), find the place in GCode and describe what was before and now -> check the message from  **fieldOfView** )
@zinenu , Thank you for helping us.

Noize4U, 2018-08-23:
It only happens when using Cura 3.4 and Marlin 1.1.9(maybe1.1.8?)I have a generic Delta/Kossel. I have not tested different Arduino/Ramps setting. When just slicing in Cura and using saved gode in printrun or octopi I do not get «click to resume»

Noize4U, 2018-08-23:
*G-code

leeb65, 2018-08-23:
@Noize4U I was going to upgrade to marlin 1.1.9 and see if it makes a difference. However marlin 2 is in Alpha or Beta and that may make a difference. Maybe even step back to marlin 1.1.7 and see if that makes a difference.

What firmware does the Ultimaker use?

@ianpaschal I am a developer too and finding these bugs is not easy, but a suggestion that may have been overlooked is no patronising. It's just a shame it's written in Python.

Noize4U, 2018-08-24:
When I printed a RAFT the printer paused (click to resume) 3 times for no apparent reason. Sometimes when starting a print (when using Cura3.4 and Marlin 1.1.9 And after printing from another program) printhead goes hiwire and starts the print 5-10 cm above the plate.

CodeOptimist, 2018-08-24:
@Noize4U I had that last part too. No idea if it's related.

maikbooehme, 2018-08-24:
hi all,

I just did a new test with marlin 1.1.9 and Cura 3.4.1 and now after the bed heating and before/during the hotend is heating I already see the "click to resume" message . 

Best regards
Maik

maikbooehme, 2018-08-24:
I have an Idea, could it be that this issue is caused by the setting of the printer when the machine setting "heated bed" was not selected, because the values for the plate temperature are not listed under material.

I'm just running an other test, where I selected the option "heated bed" .

leeb65, 2018-08-24:
@maikbooehme I didn't bother with 3.4.1, I reverted back to 3.3.1 and stayed with that and that works fine, I may upgrade the firmware on my printer to marlin 1.1.9.

On another note, I wonder if it would have the same issue with the Repetier firmware, or if it's just related to marlin.

As I already asked before: What firmware does the Ultimaker use? Is it proprietary firmware or is it Marlin.

fieldOfView, 2018-08-24:
Ultimaker uses a fork of Marlin from quite a while ago. But Ultimaker does not use USB printing anymore, which explains why USB printing is not getting any attention from Ultimaker devs.

maikbooehme, 2018-08-30:
I confirm, going back to 3.3.1 and the issue is not seen ... so for me its a code issue with in 3.4 and 3.4.x

Gazza-USA, 2018-09-14:
I see the same issue having just upgraded to Marlin 1.2.1.  I did not see this before using Cura 3.4.1 so it looks like this is an interaction issue.   I also notice that since the temp controls are now PID on Marlin 1.2.1 that the temperature is not being reported up to Cura.  So I'm assuming that Cura times out because it doesn't get any status from the Ptinter about the Temperatures.  I have files a bug report with the Marlin code guys as I suspect this is where the error is.  M105 reports correctly though.

Incidentally to fix this, if you do a PRE-HEAT the problem does not occur.  So I really suspect this is a timeout on temperature reporting.

blueeagle69, 2018-09-21:
I know this doesn't help, but I have the same. Whenever it sees a Gcode command for temperature in the start Gcode it sends a pause to the printer. I mostly use Cura, slicing to the SDcard, so a small annoyance. I'm normally stood over my printer during the first layer anyway.

Gazza-USA, 2018-09-22:
I moved to Repetier and that seems to work.  It uses the Cura engine for slicing.  Looks like it's a Cura issue.

scooper1, 2018-09-24:
i am getting this click to resume...
it happens about a average once every 20 minutes

DarkShadowsX5, 2018-09-27:
Same problem for me after recent Cura update. 
Cura 3.4.1
Printer: Prusa i3 acrylic clone( < name doesn't matter as it could be called flexible noodle i5 and have the same hardware) 
Control board: a MKS BASE v1.5 w/ RAMPS 1.4
Marlin: v1.1.4

Problem: printer heats up (click to resume) printer levels (Click to resume)
printing from SD card has a click to resume too and i just hate using sd card printing anyway. its much easier to use USB. I dont have an octopi compatible board so that's out of the question.

startup code: 
`G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000
M117 Printing...
M75 ;Start Print Timer`

i dont see any pauses.... i'm going to have to downgrade Cura and deal with the annoying "upgrade available" popup. for now.

Sam0Park, 2018-09-27:
It seems like the problem comes from combination of Cura 3.4 and some firmwares which is not reporting current temperature. 
Cura 3.4.1
Marlin 1.1.8
I just found out, If I set the initial layer temperature different than the default temperature,
it pauses in Cura's USB printing mode. 
Printing from SD card does not make the pause.
When I restored the initial layer temperature to the same as the default temperature,
Cura does not pause.

I hope we can fix this soon. 

Thanks

nallath, 2018-09-28:
We have a pull request open that should resolve it, but we haven't gotten around to testing it.

fieldOfView, 2018-09-28:
If you mean #4369, then I would say "We have a pull request open that ~~~should~~~ could - in theory - resolve it,"

The changes in that PR are fairly small, so if you feel adventurous you may help us test if it fixes this issue too: https://github.com/Ultimaker/Cura/pull/4369/files

fieldOfView, 2018-10-04:
Interesting. I have just updated another printer (BLACKBELT 3D) to Marlin 1.1.9, and now I am getting the same "Click to resume" messages, _when printing from SD!_

fieldOfView, 2018-10-05:
It looks like the problem is that Marlin does not like being polled continuously with M105 gcodes while it is preheating. During actions such as M109 and M190, Marlin reports that it is busy but Cura continues to send M105 lines. I think Marlin pauses because it has detected that commands were lost because the buffer filled up.

This should be fixed in #4487.

alekseisasin, 2018-10-05:
What I can do, make a new build(from #4487 ) and we can test together to make sure that it is solved. Do we have volunteers?

joba-1, 2018-10-05:
yes.

I’d gladly test app images on open suse leap with an anycubic kossel pulley using marlin 1.1.9.

Currently I see the confirmation request at the beginning of each print and sometimes inbetween (last 2 prints ruined, one after exactly 2.5 layers of a 5cm square, the other much later, after printing height ca. 1cm

alekseisasin, 2018-10-05:
@joba-1 , This is the link. 
https://wetransfer.com/downloads/91a7b4bcfed5347564d74ef9366e8b5620181005142210/74607f7118e23ff5eb9c3482b2cff05f20181005142210/4138be

Be aware it is **NOT STABLE** Cura build and use it only for testing purposes.

joba-1, 2018-10-05:
@alekseisasin Thanks! But unfortunately you uploaded a win64 build. As I wrote, I use open suse leap, so this build is very stable ... in not working at all :)

Sanguium, 2018-10-06:
> 
> 
> @joba-1 , This is the link.
> https://wetransfer.com/downloads/91a7b4bcfed5347564d74ef9366e8b5620181005142210/74607f7118e23ff5eb9c3482b2cff05f20181005142210/4138be
> 
> Be aware it is **NOT STABLE** Cura build and use it only for testing purposes.

So just tried this build, a bit of background:
Anet A8 - Marlin 1.1.9
Cura 3.3.1 prints fine, but temps won't update while heating.
Cura 3.4-3.5 temp still don't update, print pauses after heating when resumed prints fine.

With this 0.0.0 build temps update while heating but it still pauses after heating, and **it won't print** after resumed, it stops as soon it touches the bed and looks like it's trying to move like one step each 5 seconds, at that point I aborted the print. Tried with the same 3.5 profile I had and creating a new one from scratch, same result.

fieldOfView, 2018-10-09:
@leeb65 @neowinx and @hughsie, could you please test the build shared by @alekseisasin above?

@joba-1, just to rule out slicing issues (which we are not interested in for this build), could you slice an object with 3.4 or 3.5, save the gcode and try sending it with the private build?

joba-1, 2018-10-09:
Hi Aldo

I told it on github, but cant see my answer in this mail thread, so you
probably have not seen it:
I cannot test the private build, since it is not linux.

What I could do and I already planned to do is generate gcode on 3.5 and
print from SD and from Cura 3.3.
Also I could generate gcode on 3.3 and try to print ih with 3.5 if that
helps?

Still, testing time is limited, because Cura destroys my other serial
connections, so I cannot use it in parallel to other stuff.
-- 
iphone so i typo

fieldOfView, 2018-10-09:
@joba-1, it is a case of mistaken identity, I meant to ping @Sanguium

Sanguium, 2018-10-09:
@fieldOfView Sliced with 3.5 and gcode loaded into test build, same scenario as before, heats, pauses and doesn't print after resumed.

joba-1, 2018-10-12:
Fwiw: 

I could finally manage to build Cura myself for openSUSE linux from source (github master).
Several small prints done with this code work just fine with no pause, while the 3.5 appImage never worked. Temperature updates work during heating and during the actual print.

In case it makes a difference: I had to use another printer model as base model. In 3.5 I used DeltaBot, now I use Kossel Pro.
For some reason I could not select filament with diameter 1.75 with the DeltaBot, although I changed it in the machine settings.
This (I guess) changed the Start G-Code from

```
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm

;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
```

to

```
; info: M303 E0 S200 C8 ; Pid auto-tune 

M140 S{material_bed_temperature}; Start heating up the base
G28 ; Home to top 3 endstops
; Autolevel and adjust first layer
; Adjust this value to fit your own printer!  (positive is thicker)
; This default value is intentionally very high to accommodate the
; variety of print heads used with this printer.  Many of you will
; need tiny values like Z0 or Z0.1.  Use feeler gauges to dial this
; in as accurately as possible.
G29 Z10

; Squirt and wipe ;
M109 S220 ; Wait for the temp to hit 220
G00 X125 Y-60 Z0.1 ;
G92 E0 ;
G01 E25 F100 ; Extrude a little bit to replace oozage from auto levelling
G01 X90 Y-50 F6000 ;
G01 Z5 ;

; Set the extruder to the requested print temperature
M104 S{material_print_temperature}
```

I didn't test if that makes a difference.

The only code difference I see in the USBPrinting plugin since 3.5 is some pattern matching:

```
diff --git a/plugins/USBPrinting/USBPrinterOutputDevice.py b/plugins/USBPrinting/USBPrinterOutputDevice.py
index 4ceda5287..36c532118 100644
--- a/plugins/USBPrinting/USBPrinterOutputDevice.py
+++ b/plugins/USBPrinting/USBPrinterOutputDevice.py
@@ -326,8 +326,8 @@ class USBPrinterOutputDevice(PrinterOutputDevice):
                 if self._firmware_name is None:
                     self.sendCommand("M115")
 
-            if (b"ok " in line and b"T:" in line) or line.startswith(b"T:") or b"ok B:" in line or line.startswith(b"B:
"):  # Temperature message. 'T:' for extruder and 'B:' for bed
-                extruder_temperature_matches = re.findall(b"T(\d*): ?([\d\.]+) ?\/?([\d\.]+)?", line)
+            if re.search(b"[B|T\d*]: ?\d+\.?\d*", line):  # Temperature message. 'T:' for extruder and 'B:' for bed
+                extruder_temperature_matches = re.findall(b"T(\d*): ?(\d+\.?\d*) ?\/?(\d+\.?\d*)?", line)
                 # Update all temperature values
                 matched_extruder_nrs = []
                 for match in extruder_temperature_matches:
@@ -349,7 +349,7 @@ class USBPrinterOutputDevice(PrinterOutputDevice):
                     if match[2]:
                         extruder.updateTargetHotendTemperature(float(match[2]))
 
-                bed_temperature_matches = re.findall(b"B: ?([\d\.]+) ?\/?([\d\.]+)?", line)
+                bed_temperature_matches = re.findall(b"B: ?(\d+\.?\d*)  ?\/?(\d+\.?\d*) ?", line)
                 if bed_temperature_matches:
                     match = bed_temperature_matches[0]
                     if match[0]:
```

alekseisasin, 2018-10-15:
Right, there are not many changes in "pattern matching:". From the beginning we do not understand how it might affect this issue, you can try to replace the code from the 3.4 branch and see the difference.

fieldOfView, 2018-10-15:
> From the beginning we do not understand how it might affect this issue

What? It is not rocket science.

During some operations, such as preheating, the printer responds to new commands with "echo:busy". While it is buys, it does send temperature messages, but these are not prepended with an "ok", because the "ok" is supposed to show that a command was received and executed. So the two patches I wrote do the following:
* the pattern matching no longer looks for "ok" messages, but looks for temperature updates (this fixes the temperature updating while the printer is preheating)
* once the printer has said that it is busy, stop asking for temperature updates until the next "ok" is received (this prevents the serial buffer overflowing while preheating)

alekseisasin, 2018-10-15:
"What? It is not rocket science." I meant, how come that change does not work for other printers if there is no special change.

yodor, 2018-10-15:
I have the same issue with click to resume - delta printer with marlin 1.1.9 and cura 3.5.0-beta
Shows click to resume randomly.
My solution was to extract the AppImage files then replace the USBPrinterOutputDevice.py from tree 
cb24d58
(I did not manage to compile from master because of some issues with material not found)
All goes well for now thanks :)

fieldOfView, 2018-10-17:
It seems that #4487 may have fixed the "press to continue" after preheating, but it is still occurring during the print.

To update the temperatures in the monitor, Cura sends M105 pings every 2 seconds. It seems that if this is done during a print without waiting for an "OK" from the printer, the serial buffer on the printer may still overflow eventually (causing Marlin to complain/pause). I will update #4487 with a fix for that.

yodor, 2018-10-17:
@fieldOfView you are right, I tried a long print last night and it happened during printing atleast once.

fieldOfView, 2018-10-18:
@yodor, I have updated my branch, it would be great of you could test it again:
https://github.com/Ultimaker/Cura/blob/777470db7f06eb64d70e73468d195ab1140eba42/plugins/USBPrinting/USBPrinterOutputDevice.py

[18 more replies]

commit cb24d58a (2018-10-23): Don't flood the printer with temperature requests while says it is busy

Fixes #3994