Ultimaker/Cura #3996: [3.2.1] Segmentatioin violation on startup

sk8nfool, 2018-06-28:
**Application Version**

3.2.1 (PPA)
**Platform**

Ubuntu 18.04 (artful)
**Printer**

N/A
**Steps to Reproduce**

Start cura from command line
**Actual Results**

Crashes - outputs the following
$ cura
[(139953750898496)-MainThread] UM.Resources.__initializeStoragePaths [351]: Initializing storage paths
[(139953750898496)-MainThread] UM.Resources.__initializeStoragePaths [362]: Config storage path is /home/braun/.config/cura/3.2
[(139953750898496)-MainThread] UM.Resources.__initializeStoragePaths [370]: Data storage path is /home/braun/.local/share/cura/3.2
[(139953750898496)-MainThread] UM.Resources.__initializeStoragePaths [382]: Cache storage path is /home/braun/.cache/cura/3.2
[(139953750898496)-MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[(139953750898496)-MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [119]: Trying OpenGL context 2.0...
[(139953750898496)-MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [128]: We got at least OpenGL context 2.0: 2.1 No profile
[(139953750898496)-MainThread] UM.Qt.QtApplication.__init__ [88]: Detected most suitable OpenGL context version: 2.0 No profile
[(139953750898496)-MainThread] UM.Qt.QtApplication.__init__ [104]: Command line arguments: {'debug': False, 'trigger_early_crash': False, 'external-backend': False, 'headless': False, 'disable-textures': False, 'qmljsdebugger': None, 'file': [], 'single_instance': False}
2018-06-28 18:35:19,449 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2018-06-28 18:35:19,476 - INFO - [(139953750898496)-MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [65]: Found CuraEngine at: /usr/bin/CuraEngine
2018-06-28 18:35:19,478 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2018-06-28 18:35:19,486 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2018-06-28 18:35:19,493 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2018-06-28 18:35:19,494 - WARNING - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [200]: Plugin ConsoleLogger was already loaded
2018-06-28 18:35:19,502 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2018-06-28 18:35:19,510 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalContainerProvider
2018-06-28 18:35:19,526 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2018-06-28 18:35:19,534 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2018-06-28 18:35:19,542 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2018-06-28 18:35:19,555 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2018-06-28 18:35:19,562 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2018-06-28 18:35:19,568 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2018-06-28 18:35:19,576 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2018-06-28 18:35:19,581 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2018-06-28 18:35:19,586 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2018-06-28 18:35:19,596 - WARNING - [(139953750898496)-MainThread] STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2018-06-28 18:35:19,599 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2018-06-28 18:35:19,634 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2018-06-28 18:35:19,684 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2018-06-28 18:35:19,742 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2018-06-28 18:35:19,748 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2018-06-28 18:35:19,759 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin MonitorStage
2018-06-28 18:35:19,771 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin PrepareStage
2018-06-28 18:35:19,784 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2018-06-28 18:35:19,794 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFWriter
2018-06-28 18:35:19,800 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin ChangeLogPlugin
2018-06-28 18:35:19,804 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2018-06-28 18:35:19,811 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2018-06-28 18:35:19,815 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade30to31
2018-06-28 18:35:19,820 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2018-06-28 18:35:19,824 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2018-06-28 18:35:19,828 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2018-06-28 18:35:19,834 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2018-06-28 18:35:19,839 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2018-06-28 18:35:19,848 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2018-06-28 18:35:19,855 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2018-06-28 18:35:19,861 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileReader
2018-06-28 18:35:19,880 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2018-06-28 18:35:19,892 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimulationView
2018-06-28 18:35:19,899 - WARNING - [(139953750898496)-MainThread] UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2018-06-28 18:35:19,899 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2018-06-28 18:35:19,904 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2018-06-28 18:35:19,909 - INFO - [(139953750898496)-MainThread] UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
[... 111 lines cut ...]
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f493ffff700 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f49498fc700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/usr/lib/python3.6/threading.py", line 864 in run
  File "/usr/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f49859b7740 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 236 in initializeEngine
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 730 in run
  File "/usr/bin/cura", line 141 in <module>
Segmentation fault (core dumped)
$ 
**Expected results**

Gui to start up with first time dialogs
**Additional Information**

diegopradogesto, 2018-06-29:
Maybe is related with the PPA version of Cura. @thopiekar, do you have any clue?

thopiekar, 2018-06-29:
Hmm, He doesn't seem to use the stable PPA. It is currently using 3.3.1.
If the splash doesn't show up at all, then it seems to be a bug in PyQt5, which has been fixed due to my feedback on launchpad.net 2 or 3months before.

I would recommend to make an system update and would love to know whether he is using the PPA or not. 🙂

diegopradogesto, 2018-06-29:
OK, so let's wait for more info on that. Thanks @thopiekar for the quick response.

thopiekar, 2018-06-29:
You are welcome. Always feel free to mention me on issues 😉✌️

sk8nfool, 2018-06-29:
I'm using "deb http://ppa.launchpad.net/thopiekar/cura/ubuntu artful main"

sk8nfool, 2018-06-29:
BTW - the splash does appear but Cura crashes shortly after.

sk8nfool, 2018-06-29:
Ok - I think I misspoke - the version of Ubuntu is 17.10 (artful) - let me try my other system.

thopiekar, 2018-06-29:
I wrote with someone from Canonical about this problem and they are not going to fix it in artful or older.
Also for the future it is important to report a bug to them as soon as possible. Otherwise they can trace it back and won't fix it.

Happy slicing!