Ultimaker/Cura #3997: Octoprint PlugIn is not showing camera view in CONNECT pane

LilBub, 2018-06-29:
New PC, first time Cura install.
Installed 3.4 
Installed Octoprint Plugin
Went through the steps and no video.

After some general debugging (video works on Octoprint webserver), I referred to the recent GitHub ticket (https://github.com/Ultimaker/Cura/issues/3725#issuecomment-401177296) and downloaded and replaced that file.

Works now.

For some reason, the fix didn't get carried into the Production release?

diegopradogesto, 2018-06-29:
We are currently deploying the new OctoPrint cura package, and luckily today you will find the new version 3.4.1 provided by @fieldOfView in the new Toolbox. In fact this version was specifically created to fix this issue with the camera.

Vandrasc, 2018-06-29:
This should be now fixed.

closed 2018-06-29 by Vandrasc

fieldOfView, 2018-06-29:
Toolbox still seems to serve up OctoPrintPlugin 3.4.0. This was fixed in OctoPrintPlugin 3.4.1

diegopradogesto, 2018-06-29:
Still waiting for somebody to merge our staging branch into production.

LilBub, 2018-06-29:
I am not seeing 3.4.1 on the download page?  Unless this is rebundling 3.4?

diegopradogesto, 2018-06-29:
It should be now available for download in the toolbox.

LilBub, 2018-06-30:
Will test.